magicbug/Cloudlog #2025: Serial don't save with space

Stiles96, 2023-03-05:
**Describe the bug**
Received Serials in Contest Mode won't be saved, if you are using space to jump the next field. Missing a trim() before convert?

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Contest Log
2. Log with serial
3. Jump to next field by using the space
4. Save QSO (Enter doesn't work)

**Expected behaviour**
Serial will be saved.

**Screenshots**
Missing received Serial:
[image]

**Desktop (please complete the following information):**
 - OS: Win 10
 - Browser Firefox
 - Version 110

AndreasK79, 2023-04-01:
Enter was only triggered when in received exchange. Guess it was never expanded when adding more exchanges. I've pushed a change to the dev branch so that either serial or gridsquare received, logs qso when hitting enter.

closed 2023-04-03 by AndreasK79

Stiles96, 2023-04-03:
The Issue isn't fixed. The issue is the space at the end of the serial. You can save the QSO, but without the serial. (Step 3, You see the space also in the screenshot).

AndreasK79, 2023-04-03:
Can't reproduce that problem here. Received serial saves here after hitting space in serial, then entering grid and hitting enter (or clicking button).

AndreasK79, 2023-04-03:
It might be a problem related to differences in database setup, and how strict things are.

Can you edit the source code and try to add a trim like this?

logbook_model.php, line 23:
      `$srx = $this->input->post('exch_serial_r') == '' ? null : trim($this->input->post('exch_serial_r'));`

Stiles96, 2023-04-03:
I tested and it doesn't work.
I debugged also the session and can see that the serial isn't transmitted by Firefox itself:
[image]

In Chrome it works very well.

AndreasK79, 2023-04-03:
Yes, Firefox seems to not work all the time. I got it to not save the serial received, but some times it saved it as well. I have no clue on what is going on regarding this, so this is not something I can fix.

AndreasK79, 2023-04-05:
I had a discussion with @phl0, and it seems you were right regarding the space. In exchange, space is not written, but this was not implemented in the received serial. I pushed a fix that implements this. Seems to fix the issue you reported.

phl0, 2023-04-05:
Tested and fix confirmed on my end. Pls test and report :)

Stiles96, 2023-04-05:
It works! Good work.