magicbug/Cloudlog #2030: Error after update

DL3EL, 2023-03-10:
we did an update to the latest version with git pull and after updating config.php a start of cloudlog throws the follwoing error:

A Database Error Occurred
Error Number: 1062

Duplicate entry 'version2_trigger' for key 'option_name'
INSERT INTO `options` (`autoload`, `option_name`, `option_value`) VALUES ('yes','version2_trigger','false')
Filename: migrations/102_add_version_two_trigger_to_options.php
Line Number: 18
What are we doing wrong?

vy 73 de Thomas, DL3EL & Patrick, DL5ZR

AndreasK79, 2023-03-15:
Hi, you didn't specify which branch you are using, but I have pushed a fix for the migrations scripts that should fix this.
This is the dev branch, so you need to use that one.

DL3EL, 2023-03-15:
HI Andreas,
it was the master branch. We will try the dev branch, thanks.
vy 73 de Thomas, DL3EL

AndreasK79, 2023-03-18:
@DL3EL did it work?

DL3EL, 2023-03-18:
@AndreasK79 sorry, I was travelling und have just arrived back home. I hope, I can look at it tomorrow

AndreasK79, 2023-03-18:
No problem. I think it should work. I could replicate it here, and fixed the scripts that gave the errors.

DL3EL, 2023-03-19:
Hi Andreas,
this problem has been resolved. Now we are back to the originla proble, that there are no logbooks after the update. The hint to change something on db level does not work, as the pw is gone :-( So next step is probably to rebuild the system completely, we have a backup of the log data.

phl0, 2023-03-19:
There is an option to reset the SQL passwords if needed. It is a little tricky but it works. But it should still be in application/config/database.php? This file is not touched no updates ...

DL3EL, 2023-03-19:
cool @phl0 , file untouched, pw still there and login works. Now we can try again. You know the db and mysql far better than I do, can you quick send me the two or three commands I have to do to achieve "...only real method is to manually assign the location via the table in the database this is the station_profile table and you want to edit the user_id column". That is what Peter sent me as asolution for issue #1920.
Thanks, Thomas

phl0, 2023-03-20:
First you want to make a backup of your tables (I do this using mysqldump).

Then you would need to lookup the user id you want the station locations to:

```
use cloudlog;
select user_id, user_name from users where 1;
```

Then assign the/all station profiles to the user(_id) selected from the first query:

```
update station_profile set user_id = 6 where 1;
```

This assigns *ALL* station locations to the user with user_id 6.

closed 2023-03-20 by DL3EL