magicbug/Cloudlog #2047: 504 gateway

iplayradionetworks, 2023-03-22:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

magicbug, 2023-03-27:
As this is an issue with server config i'd suggest opening a discussion.

closed 2023-03-27 by magicbug

iplayradionetworks, 2023-03-28:
Hi .

How is that a server issue ?

I have checked everything on my side but nothing wrong . 

I have even increased the timeout length in php . 

Can you suggest me where to check ?

Best 73s
Jimmy

phl0, 2023-03-28:
I guess it is using a kind of proxy in front of the Cloudlog instance? Then the timeouts apply to the proxy ...

iplayradionetworks, 2023-03-28:
No there is no proxy. 

Just plain apache . 

Should I be checking for apache timeout sessions In case to increase that parameter too ?

phl0, 2023-03-28:
Strange that it does output a 504 gateway timeout then. This is normally happening if a reverse proxy cannot access the backend. Is apache used as reverse proxy?

iplayradionetworks, 2023-03-28:
> Strange that it does output a 504 gateway timeout then. This is normally happening if a reverse proxy cannot access the backend. Is apache used as reverse proxy?

No . No reverse proxy with apache.

So I guess I must discover what it can be

phl0, 2023-03-28:
Answered in https://github.com/magicbug/Cloudlog/discussions/2049#discussioncomment-5451607.

iplayradionetworks, 2023-04-04:
Here is what I get after uploading.....
[image]