magicbug/Cloudlog #2055: [Feature Request] Configure default value or "Power"

m0urs, 2023-04-05:
I would like to have the value for the transmit power as a config setting in a station location. So whenever a new QO is created for this location, this value should be the preset value (which you would be able to overwrite).

Currently after I start a new Cloudlog session I need to manually set the value every time (and often forget to set it). If you have set the value and saved at least one QSO, the value will be remembered already for the rest of the session, which is ok.

Unfortunately I still have not really understand how the programming model of Cloudlog works, so I can sometimes do small changes by myself, but others I am not able to do :-(

phl0, 2023-04-27:
At https://github.com/magicbug/Cloudlog/pull/2091 there is a PR that we would love to hear your feedback on. ;-)

m0urs, 2023-04-28:
Hi Florian, that looks fine to me. I was able to define my 10W into all my locations and it appears for new QSOs now.  Thanks a lot!

phl0, 2023-04-28:
Anyone else to test here before I undraft this PR? @AndreasK79 @magicbug?

AndreasK79, 2023-04-28:
Tested ok here.

closed 2023-04-28 by AndreasK79