magicbug/Cloudlog #2059: [Feature Request]

techfixpros, 2023-04-08:
Please provide a function to automatically import and attach QSL cards downloaded from eQSL via DXKeeper.

DXKeeper from DXLabs can automatically download eQSL cards, and stores them in the format <callsign>_<band>_<mode>_<yyyy>_<mm>_<dd>_<hh>_<mm>_<ss>.png. 

The files can be dropped in a folder accessible from cloudLog and matched to the confirmed QSO, and automatically added for viewing.

magicbug, 2023-04-08:
Cloudlog already will download the eQSL card click the down arrow in the logbook next to eQSL and its saved locally so this would just need a script that would read the file name and assign it correctly into the database table.

phl0, 2023-05-09:
> so this would just need a script that would read the file name and assign it correctly into the database table.

Something like this?

[image]

techfixpros, 2023-05-09:
Yes, that and the backend to automatically download it and link the stored file to the qso for local storage.

techfixpros, 2023-06-04:
This is resolved. Thank you for the new feature.

closed 2023-06-04 by techfixpros