magicbug/Cloudlog #2060: Feature request: CAT widget on dashboard should refresh on changing frequency of rig

PD0RZH, 2023-04-09:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
CAT widget on dashboard should refresh on changing frequency of rig

**To Reproduce**
Steps to reproduce the behaviour:
Setup CloudlogCat first
1. Go to /index.php/radio
2. Add hardware interface
3. On success > Change frequency of rig
4. On /index.php/radio frequency change is displayed
5. On /index.php/dashboard the change of frequency Is not displayed

**Expected behaviour**
CAT widget on dashboard should refresh on changing frequency of rig

**Additional context**
**Additional request....**
It would also be fine if the CAT widget could be displayed on different sources like a homepage or QRZ.com page for example. HTML code for that should be made available. That way someone can easily find your whereabouts on the frequencies.

**Extra, extra read all about it.....**
Haven't checked it but is the CAT widget available on the Public page of Cloudlog?

Michael-AUG, 2023-05-24:
Agreed - this would be really good please :)

magicbug, 2023-05-28:
This is now added to the dev branch the table updates every 5 seconds

closed 2023-05-28 by magicbug