magicbug/Cloudlog #2071: Blank page after installation wizard

ALJAZHERO, 2023-04-23:
Hi.
I got the installation working up to the installation wizard page. when I fill out all of the required information and click Install, I get a blank page loaded up. Do you have any sugestios on fixing this up? 
Thanks in advanced.

ALJAZHERO, 2023-04-24:
[image]
[image]
[image]

Yes, here they are.

magicbug, 2023-04-24:
I'm going on a guess but I'll go with that it failed to complete the MySQL insert you might get more info looking at the server log files

magicbug, 2023-05-05:
2 weeks without a reply so closing

closed 2023-05-05 by magicbug

minorsecond, 2023-07-11:
I'm having this issue too. After running the installation wizard (a few times):

```
MariaDB [cloudlog]> show tables;
Empty set (0.000 sec)
```
`Jul 11 19:11:03 log mariadbd[1656]: 2023-07-11 19:11:03 33 [Warning] Aborted connection 33 to db: 'cloudlog' user: '******' host: 'localhost' (Unknown error)`