magicbug/Cloudlog #2072: Check for QSOs without station_id is not working

phl0, 2023-04-24:
application/Logbooks_model.php line 3646:

```
$this->db->where('station_id =', 'NULL');
```

results in SQL command:

```
SELECT * FROM `TABLE_HRD_CONTACTS_V01` WHERE `station_id` = 'NULL' 
```

This cannot work because station_id is set up as int NULL DEFAULT NULL. So 'NULL' as text value can never be assigned. I guess the intended use was to find QSOs without a station id (i.e. station_id = NULL). So I guess this query should be

```
$this->db->where('station_id =', NULL);
```

This actually finds QSOs with absent station_id.

@AndreasK79 Could you please verify?

AndreasK79, 2023-04-24:
That will not work. Looking for text value in int column is no good. Nice find.

phl0, 2023-04-24:
Fixed with https://github.com/magicbug/Cloudlog/pull/2073.