magicbug/Cloudlog #2075: Exception: Unknown column 'radio' in 'where clause'

pa8s, 2023-04-24:
**Describe the bug**
Exception: Unknown column 'radio' in 'where clause'

**To Reproduce**
Unable to recall performed steps :(

**Expected behaviour**
No error in logfile

**Logfile**
ERROR - 2023-04-24 21:04:55 --> Severity: error --> Exception: Unknown column 'radio' in 'where clause' /mnt/www/system/database/drivers/mysqli/mysqli_driver.php 301

**Server**
OS: FreeBSD 13.1
Server: nginx/1.24.0
PHP: 8.1.18
Codeigniter: 3.2.0-dev
Git branch: dev
Git commit: https://github.com/magicbug/Cloudlog/commit/d6139439734f92fbd81cfcb85c3a616fff322832
Git Tag: https://github.com/magicbug/Cloudlog/releases/tag/2.3.1

pa8s, 2023-04-25:
Ah clear, I just couldn't recall the situation but I guess it happened when I changed frequency or mode. Anyhow, just checked and the radio column does exist in the cat cable and contains 3 rows.

phl0, 2023-04-30:
So could you close this one then? Or did the issue arise again?

closed 2023-04-30 by magicbug

pa8s, 2023-04-30:
Pulled the latest dev, checked a moment ago and the error was not seen in the log file.