magicbug/Cloudlog #2081: Live/Post-QSO Save-Button does not work

DrDerrek, 2023-04-25:
**Describe the bug**
When logging a QSO live or post-QSO, the "Save QSO" button does not work. It is mising its JavaScript call

**To Reproduce**
Just enter "Live QSO" and try to save a QSO

**Expected behaviour**
Saving the entered QSO

**Logfile**
There is no entry in the console or Logfile

**Server**
OS: Debian 11 (bullseye)
Server: apache2 (2.4.56-1~deb11u1)
PHP: 7.4.33-1+deb11u3
Git branch: dev
https://github.com/magicbug/Cloudlog/commit/8654e9e15e368d10935142bcebcdedfe508f98b9
Git Tag: https://github.com/magicbug/Cloudlog/releases/tag/2.4.1

phl0, 2023-04-25:
Is the DXCC field filled?

magicbug, 2023-04-25:
> Is the DXCC field filled?

Beat me to it, what callsign? should really put "none" if nothings found to make things easier

phl0, 2023-04-25:
I guess he did not load/update the DXCC data (after installation)?

closed 2023-04-25 by DrDerrek

phl0, 2023-04-25:
> I checked the installation guide as well as the hints in the documentation. There is no notice you need to do that anywhere. 

What about here (see substeps):

https://github.com/magicbug/Cloudlog/wiki/Installation#5-run-the-cloudlog-install-wizard