magicbug/Cloudlog #2086: Contest Logging: Serial number counts up if you make a change in "Exchange Type"

m0urs, 2023-04-27:
Every time you change the value in "Exchange Type" in the contest logging, the serial number will be inceremented by one.

Steps to reproduce:

1. Open contest logging
2. Reset contest log session
3. Change "Exchange Type" to "Serial"
4. You see an empty "Serial" field which is correct
5. Change "Exchange Type" to "Serial plus Exchange"
6. You see the value "1" in "Serial" field which in this case coud also still be ok
7. Now just change the "Exchange Type" to something else. Every time  you do that, the "Serial" field increments by one.

AndreasK79, 2023-04-28:
@m0urs thanks for reporting. I've pushed a fix to dev now. As far as the tests I did, it fixed the problem.
Would appreciate if you tested and verified.

AndreasK79, 2023-04-28:
@m0urs thanks for testing. Setting serial several times and overwriting is not good. Pushed another fix.

m0urs, 2023-04-28:
I now have the issue again, that after I logged a QSO with e.g. exchange type "Serial" and then change the exchange type and leave the contest logging view, that afterwards it increments every time I change the exchange type. So quite the same issue as in the beginning.

m0urs, 2023-04-28:
Now the issue happens not while changing the exchange type but every time I leave the logging view and then return again ...

AndreasK79, 2023-04-28:
Thanks, guess it's back to the drawing board here. I was certain that I had it fixed when I made the last commit, but probably not checking the correct things.

m0urs, 2023-04-28:
Let me know if I should test another fix then!

AndreasK79, 2023-04-29:
@m0urs I hope I nailed it this time. Please have another go and report back.

m0urs, 2023-04-30:
Yes, I would say, now it is working as expected. I tested several different cases and every time the serial number had the value which I would have expected. Thanks for fixing!

closed 2023-04-30 by m0urs

AndreasK79, 2023-04-30:
Thank you for testing. It is always good to have someone test out things. Often you make stuff a certain way without thinking about every possible thing.