magicbug/Cloudlog #2113: eSL images not downloaded

m0urs, 2023-05-07:
I just saw, by accident, that I have only 116 images "images/eqsl_card_images" and for most of my eQSL confirmations I cannot see the QSL image. I only get the information 

[image]

My last image is from April 8th and I do have eQSL confirmations after that date which have no image as well.

So I have the following questions:

- Should an image be downloaded for every eQSL confirmation?
- If yes, what could be the reason that no images are downloaded any longer?
- I would have said that I had more images already in that directory in the past (although I am not 100% sure) so maybe something has been lost in the past. Is there any possibility to re-download all eQSL images for all existing, confirmed QSOs?

AndreasK79, 2023-05-07:
From what I can figure out looking in the code, the image is not automatically downloaded. It is only downloaded when you click on the confirmation.

I read in the code that it is also rate limited, so that using a script for downloading all does not seem like a good idea.

Perhaps someone with more knowledge about eQSL have some more information.

m0urs, 2023-05-07:
Ah ok, thank you. If it is downloaded if I click on the confirmation, then this would be good enough. However, it seems that this is broken at the moment. So I only get the error above and the image is not displayed.

phl0, 2023-05-07:
Jepp. eQSL is rate limited. Bulk downloading will not work. And yes the code was designed to only download on demand (maybe due to rate limit). 
On my end eQSL download works at the moment. Maybe temporary issue?

magicbug, 2023-05-07:
Correct eqsl rate limit and that's why I coded it to do them on demand not in bulk.

This issue will resolve itself after a few minutes

closed 2023-05-07 by magicbug

m0urs, 2023-05-07:
It does not work for me with the current DEV branch. If I switch to the current master then it is working again. @AndreasK79 , could it be that some of your code regarding eQSL sync from command line has broken the eQSL image display?

m0urs, 2023-05-07:
The following errors are logged while trying to display the eQSL image:

```
ERROR - 2023-05-07 15:41:36 --> Severity: Warning --> Undefined property: eqsl::$user_model /var/www/cloudlog.urspringer.de/html/application/controllers/Eqsl.php 485
ERROR - 2023-05-07 15:41:36 --> Severity: error --> Exception: Call to a member function get_by_id() on null /var/www/cloudlog.urspringer.de/html/application/controllers/Eqsl.php 485
```

m0urs, 2023-05-07:
@AndreasK79  It seems to work again if I add the following lines to "application/controllers/Eqsl.php":

**// Check logged in
$this->load->model('user_model');**

in line 473 like:

```
        function image($id) {
                $this->load->library('electronicqsl');
                $this->load->model('Eqsl_images');

                // Check logged in
                $this->load->model('user_model');

               if($this->Eqsl_images->get_image($id) == "No Image") {
		
```

phl0, 2023-05-07:
This fixes it: https://github.com/AndreasK79/Cloudlog/pull/7.

AndreasK79, 2023-05-07:
Thanks @m0urs and @phl0. The fix is merged.

m0urs, 2023-05-07:
And it does work for me 👍

AndreasK79, 2023-05-07:
@m0urs great. I think most of the stuff should be sorted, but you never know.