magicbug/Cloudlog #2116: Error when trying to load new eQSL images

ofadam, 2023-05-08:
With the latest dev release, I'm unable to request any new eQSL images – I get the following error 100% of the time:

"The requested content cannot be loaded. Please try again later."

Here's the actual server error underneath:

A PHP Error was encountered

Severity: Warning

Message: Undefined property: eqsl::$user_model

Filename: controllers/Eqsl.php

Line Number: 487
An uncaught Exception was encountered

Type: Error

Message: Call to a member function get_by_id() on null

SERVER DETAILS:
- Apache/2.4.54 (Ubuntu)
- PHP Version 8.1.18

phl0, 2023-05-08:
This is known and we already have a fix. Will be in dev soon. Thank you for reporting.

AndreasK79, 2023-05-08:
This is fixed in dev.

closed 2023-05-08 by AndreasK79