magicbug/Cloudlog #2121: Feature request: Add "About" or "Version" to Account drop-down (top right corner) when running on a Release branch

gx1400, 2023-05-08:
When running on a Release (https://github.com/magicbug/Cloudlog/releases), e.g. version 2.4.2, it would be nice to have a convenient place in Cloudlog to find which release you're running on.  This may exist and I don't see it quickly.

This version/about page could also include a "Current Version:" and "Newest Release:" verbiage.

It would also be nice if there were a dismiss-able banner that would pop up when there is a newer release than you are currently running.

I am not familiar with the Git Release API, but quick google found this: https://docs.github.com/en/rest/releases/releases?apiVersion=2022-11-28

gx1400, 2023-05-08:
https://api.github.com/repos/magicbug/Cloudlog/releases

magicbug, 2023-05-08:
Already exists in /debug it was only available in developer mode but just fixed that in the dev branch so that its shown to anyone with the Admin role

magicbug, 2023-05-08:
[image]

magicbug, 2023-05-08:
Currently have no intentions to change branch names at this point.

closed 2023-05-08 by magicbug