magicbug/Cloudlog #2141: QSO End Date not updated when uploading new LoTW certificate

pa8s, 2023-05-14:
**Describe the bug**
My previous LoTW certificate was valid until 12 May 2023. A renewed certificate was issued 14 April 2023 and subsequently uploaded to my Cloudlog instance. Up until 12 May the scheduled upload via cron went smoothly. Since then no upload was performed via cron as via manual syncing within Cloudlog. To check if it was certificate releated I uploaded a few logs of QSOs I've made on the 13th via TrustedQSL and this worked as expected. 
So it might be date related? As I have log entries later than 12 May in my logbook I was surprised to see the certificate list in Cloudlog was still showing 'QSO End Date' as 12 May 2023 although the last QSO was made on the 13th. So that date has nothing to do with log entries. To check I manually edited the date in the sql database to a later date, this changed the date in the certificate list and did a manual sync. Now the last QSOs are uploaded to LoTW, checked by logging in at ARRL. 

**To Reproduce**
Try to upload/sync logbook with QSO dated later than the 'QSO End Date' in the certificate list. This should fail.

**Expected behaviour**
When uploading/updating certificate the 'QSO End Date' is updated to what the certificates states. In my case the 'QSO End Date' is set to 'N/A' according to TQSL.
If a 'N/A' date can not be handled by Cloudlog, at least set the 'QSO End Date' to the expiry date of the certificate. 

**Logfile**
N/A

**Server**
OS: FreeBSD 13.1
Server: nginx/1.24.0
PHP: 8.1.18
Codeigniter: 3.2.0-dev
Git branch: dev
Git commit: https://github.com/magicbug/Cloudlog/commit/b52449da3c0bfc87f2f22ce2bef92acba4431023
Git Tag: https://github.com/magicbug/Cloudlog/releases/tag/2.3.1

magicbug, 2023-05-14:
stumbled into this bug the other day, solution for now is to delete the cert from Cloudlog and reupload it, however, will get it resolved.

@phl0 oh look the issue i had!

phl0, 2023-05-14:
I will have a look.

phl0, 2023-05-21:
@ryandzink This is the open issue for this. Did you re-upload the same cert? And has the previous one been uploaded before August 2022?

ryandzink, 2023-05-21:
@phl0 yes I uploaded the same cert file as was previously imported into my instance. The last time it had been uploaded was August 29 2022

phl0, 2023-05-21:
> @phl0 yes I uploaded the same cert file as was previously imported into my instance. The last time it had been uploaded was August 29 2022

Ok thats a good hint as about that time I introduced the QSO date stuff. Maybe I made a stupid mistake here. Will check. Does it now show the correct date?

ryandzink, 2023-05-21:
Yeah, here's a before and after of the same file uploaded:

[image]

[image]

phl0, 2023-05-21:
> Yeah, here's a before and after of the same file uploaded:

Weird. It shows the same dates ... That is gonna be a tricky one ... :)

ryandzink, 2023-05-21:
Oh good grief I screwed up my upload... too my screenshots!

Before: 
[image]

After:
[image]

phl0, 2023-05-21:
> Oh good grief I screwed up my upload... too my screenshots!

Ok. That looks ... different now ... I guess I made a mistake in the migration for existing certificates. I will dig into this. Thank you for the evidences.

phl0, 2023-05-21:
Note to myself: Corresponding PR is/was https://github.com/magicbug/Cloudlog/pull/1580.

pa8s, 2023-05-21:
Checked with latest dev and 'QSO End Date' is now set to one day prior to expiry date of certificate. Thanks for the fix.

phl0, 2023-05-21:
> Checked with latest dev and 'QSO End Date' is now set to one day prior to expiry date of certificate. Thanks for the fix.

It should not be set to one day prior to expiry of the certificate in general. The QSO validity dates are separate properties of the certs which you can set during creation of the cert. But for most use cases it is the same as validity period of the cert itself probably ...

phl0, 2023-05-21:
Btw could you please close here as the fix obviously works :) tnx

closed 2023-05-21 by magicbug