magicbug/Cloudlog #2152: QSOs are not detected as needing to be uploaded to LoTW

ryandzink, 2023-05-21:
**Describe the bug**
QSOs across all of my station locations created after a certain date (around May 5 this year) no longer are getting picked up when trying to sync to LoTW.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Logbook of the World sync page
2. Click manual sync

**Expected behaviour**
The station locations that have QSOs since May 5 to current should all have .tq8 files generated and uploaded, however they do not and all say No QSOs to Upload.

**Screenshots**
[image]

I am not clear what the QSO End Date column means, maybe just the date of the latest QSO uploaded with the certificate?
[image]

**Additional context**
It looks like the issue is with the dates in the query `get_lotw_qsos_to_upload` is executing. I can correctly query QSOs needing to be uploaded with 

```sql
SELECT COL_PRIMARY_KEY,COL_CALL, COL_BAND, COL_BAND_RX, COL_TIME_ON, COL_RST_RCVD, COL_RST_SENT, COL_MODE, COL_SUBMODE, COL_FREQ, COL_FREQ_RX, COL_GRIDSQUARE, COL_SAT_NAME, COL_PROP_MODE, COL_LOTW_QSL_SENT, station_id FROM Cloudlog.TABLE_HRD_CONTACTS_V01
WHERE station_id = 1
AND (COL_LOTW_QSL_SENT IS NULL 
OR COL_LOTW_QSL_SENT != 'Y')
AND COL_PROP_MODE != "INTERNET"
ORDER BY COL_TIME_ON DESC
```

so I am not sure which dates are getting populated for the date filter that are making the query return no QSOs suddenly.

[image]

phl0, 2023-05-21:
Might relate to https://github.com/magicbug/Cloudlog/issues/2141.

ryandzink, 2023-05-21:
@phl0 this certificate has been here a while (6+ months for sure), it sounds like a potential fix though is to remove and re-add it?

magicbug, 2023-05-21:
Delete cert from Cloudlog and add back

ryandzink, 2023-05-21:
OK, that seems to have worked and I see several .tq8 files being generated... 

[image]

But why did the old cert stop working? I see now the cert validation happens prior to even the SQL query running, so I get why nothing was getting generated. Why was the QSO end date on the cert set to a random date this month? After upload the new certificate shows a QSO end date of the day just prior to the cert expiring and it stays that way, will that change over time somehow?

[image]

magicbug, 2023-05-21:
Will close as its already an open issue, not quite sure why but looking into it!

closed 2023-05-21 by magicbug