magicbug/Cloudlog #2162: QSOs not showing up in logbook

minorsecond, 2023-05-23:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
When saving a live or post QSO, it does not appear in the logbook.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to QSO - live or post
2. Enter data and save
3. Go to logbook
4. See no new entries

**Expected behaviour**
The latest QSOs should appear on the logbook page.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Windows 11
 - Browser [e.g. chrome, safari] Edge

**Additional context**
I've checked my station locations the correct one is active.

magicbug, 2023-05-23:
In Station Logbooks have you created a logbook and associated the correct locations with it?

minorsecond, 2023-05-23:
Yes, I have one logbook and 4 different locations linked to it,
including the active location.

magicbug, 2023-05-23:
If you can provide some screenshots that would be useful.

If this is a new install you could try updating the Country Files under admin but I cant see that resolving this issue, but I don't believe there's a bug anywhere.

minorsecond, 2023-05-23:
Sure. This installation is a few years old. I haven't been active on the radio for a few months but noticed some time ago that this was happening. I just updated it and it's still going on.

[image]
[image]
[image]

minorsecond, 2023-05-23:
The top location in the location page is the active location in the logbook.

phl0, 2023-05-23:
Has the installation been updated lately? Is there any error message? Any hints in the dev tools of the browser? And if access to the logs any hints there?

minorsecond, 2023-05-23:
I updated this morning, but the issue happened both before and after doing so. No error message from the UI.

The only message I see in the console is one that says it couldn't load content for bootstrap-dialog.css.map.

phl0, 2023-05-24:
Do you have access to the logs of the webserver and/or Cloudlog?

minorsecond, 2023-05-24:
I see the `logs` directory in `/var/www/html/application/logs` with an `index.html` file in it, but no actual log files.

There's not much in  `/var/log/apache2/error.log` or `access.log`. Just a line about the server restarting.

minorsecond, 2023-06-02:
After the latest update, QSOs are showing up again as expected. However, the ones that didn't show up previously are still missing. Is there a way to recover these by chance? No worries if not, as there aren't a large amount.

closed 2023-07-08 by magicbug