magicbug/Cloudlog #2185: Feature Request: Install Cloudlog with YunoHost

mattbk, 2023-06-01:
**Describe the bug**
Not a bug, but feature request to add installation with YunoHost: https://yunohost.org/.

Intro to packaging is here: https://yunohost.org/en/packaging_apps.

I have opened a PR to add Cloudlog to the YunoHost app wishlist, so this issue is to close the loop. https://github.com/YunoHost/doc/pull/2274.

**Expected behaviour**
Search the YunoHost app directory and install Cloudlog from there.

magicbug, 2023-06-01:
I understand that some may desire this, but it may not be worth the time that could be dedicated to Cloudlog.

mattbk, 2023-06-01:
I get that. Mostly I wanted to get an issue out there im case others might be searching for it.

magicbug, 2023-07-09:
This appears to have been implimented https://github.com/YunoHost-Apps/cloudlog_ynh if you can confirm

mattbk, 2023-07-10:
Awesome. I was able to install using that package. Thanks for the note!

closed 2023-07-10 by magicbug