magicbug/Cloudlog #2189: EQSL card does not display with multiple station locations

PD0RZH, 2023-06-05:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
EQSL does not display when using multiple locations

**To Reproduce**
Steps to reproduce the behaviour:
1. Create a profile for a new location at EQSL.cc Make note of the EQSL Nickname
2. Under station locations in Cloudlog make a location and set the eQSL QTH Nickname
3. Log A QSO under this location in Cloudlog and upload to Eqsl using EQSL export
4. Download the QSO from Eqsl using the Cloudlog import from EQSL tool
5. Confirm that the QSO has been updated with the EQSL information
6. Click the green down arrow on the Dashboard overview next to the QSO that has been confirmed.
7. The image won't display. Error message : The requested content cannot be loaded. Please try again later. 
8. Unfortunately later nothing happens either
9. Confirmed QSO;s on my HOME QTH logbook will display an image by the way

**Expected behaviour**
Image should display

**Screenshots**
The download link refers to cloudlog/index.php/eqsl/image/ folder with a number

**Desktop (please complete the following information):**
 - OS: Mac OS
 - Browser Safari
 - Version 16.4 (18615.1.26.110.1)

int2001, 2023-06-23:
same problem here.

int2001, 2023-06-27:
i think this issue can be closed

PD0RZH, 2023-06-29:
Can confirm it works as expected

closed 2023-06-29 by PD0RZH

Werzi2001, 2023-07-03:
Since this update I have problems receiving my eQSL cards. It is correct that before that I had to change the eQSL account name between with and without /P to make it work (depending on the station I wanted to update). Anyhow now I can't get it to work at all for my portable stations.
I have /P set in the callsign of the station but that doesn't seem to have been stored in the QSO correctly (`COL_STATION_CALLSIGN` is without /P). That probably is not the fault of Cloudlog but the software I use to send the QSO to Cloudlog. Anyhow it is not possible to fix that without doing it in the database directly as `COL_STATION_CALLSIGN` is not visible in the UI and can't be changed there.
I think it would be better to not use `COL_STATION_CALLSIGN` in the QSO but to use the station callsign in the associated station. What do you think?

phl0, 2023-07-03:
That is going to be a general issue as we have several data belonging to the station profile stored in fields in the QSO table. That means if you change the station the QSO data re station callsign etc. will be invalidated because you cannot update those files by hand.