magicbug/Cloudlog #2198: Stop syncing to qrz.com if Key is expired

int2001, 2023-06-14:
if one enters a qrz.com-api-key, cloudlog syncs to qrz.com (Great Feature!)
But if the qrz-subscription expires, cloudlog will try the sync **for each (unsynced) QSO** every time.

Suggestion: If the first Request failed with the following Errormessage, then stop trying to upload the rest of the QSOs.

`--> QRZ upload failed with the following message: STATUS=AUTH&REASON=user does not have a valid QRZ subscription to use this function`

Saves a lot of traffic.

magicbug, 2023-06-14:
This seems like a good reason, if that's returned simple solution might be to delete the API key

int2001, 2023-06-14:
Yes, one can do that. But the "normal User" isn't aware of that. He's confused about "no sync" while the server is establishing more and more connections.
Extreme situation: Fresh Account. User imports everything via ADIF-Import und has a key. But key is invalid.

So CL tries every croncycle to sync (happened here: More than 200 MB every hour)

int2001, 2023-07-05:
Fix with PR: https://github.com/magicbug/Cloudlog/pull/2252

closed 2023-07-05 by magicbug