magicbug/Cloudlog #2207: php and database error 1064

TheMockTv, 2023-06-17:
I go to qso live or post it doesn't matter and enter my qso with callsinge, mode, band, location, and in Gridsquare I enter the locator, for example JO81IS, then press save and get a php error

A PHP error was encountered
Severity: Warning

Message: Attempt to read property "station_gridsquare" on null

File name: interface_assets/footer.php

Line Numbers: 943 and 947

have the log on a ct container with ubuntu 20.04 lts

Apache/2.4.41 with php7.4 and 8.0 the same Problem
mariadb 10.3.38-MariaDB-0ubuntu0.20.04.1

as described the folder has apache rights so www-data owner and group on 777 everything

mi 12 pro
android version 13
chrome app
version 114.0.5735.131

windows 10 pro
opera
Version:99.0.4788.65

or
edge
Version 114.0.1823.51

the same

how can I solve the problem
as it is now i can't enter any qso's because he can't save them through the gridsquare

[screenshot 1](https://pcore.de/n7pidr7z)
[screenshot 2](https://pcore.de/xh39gicd)

magicbug, 2023-06-17:
In Station Location being used did you set a grid square error is hinting you didn't

TheMockTv, 2023-06-17:
Okay, I didn't have a station location yet

How about the radio, where can I enter it?

Do I also need to create a new station location for Portable or what does that look like?

but he still doesn't want to save my QSO although I've created a station now

Excuse me, I passed my DO license in Germany yesterday and this is all new territory for me with the logbook

73 DO4NRW

TheMockTv, 2023-06-17:
Whatever the reason, it suddenly works, thanks for the help and have a nice weekend

closed 2023-06-17 by TheMockTv