magicbug/Cloudlog #2208: Custom Maps no longer working

m0urs, 2023-06-18:
For some time  the Analytics -> Custom Maps feature is no longer working for me. No pins are displayed in the map, regardless which time I enter.

In the browser developer tools I see a 403/Forbidden error for this URL as soon as I click on Load Map:

[image]

Is this working for all others?

I updated to the latest Dev but the issue is still there.

I also see the following errors in my Apache log:

```
[Sun Jun 18 18:31:05.783494 2023] [rewrite:error] [pid 52965:tid 139984271283968] [client 176.xx.xx.xx:40895] AH10411: Rewritten query string contains control characters or spaces, referer
[Sun Jun 18 18:31:18.430677 2023] [rewrite:error] [pid 52966:tid 139984304854784] [client 176.xx.xx.xx:63086] AH10411: Rewritten query string contains control characters or spaces, referer
[Sun Jun 18 18:32:03.422706 2023] [rewrite:error] [pid 52966:tid 139984304854784] [client 176.xx.xx.xx:32647] AH10411: Rewritten query string contains control characters or spaces, referer

```

I do have the index.php rewrite activated:

```
RewriteEngine On

RewriteCond %{REQUEST_URI} ^/(backup)/$
RewriteRule ^(.*)$ /index.php?/$1 [L]

RewriteCond %{REQUEST_FILENAME} !-f
RewriteCond %{REQUEST_FILENAME} !-d
RewriteRule ^(.*)$ /index.php?/$1 [L]

```

magicbug, 2023-06-18:
As i dont run htaccess on mine not sure, have you tested your install without that enabled

m0urs, 2023-06-18:
Yes, if I change the config.php to use the index.php again and remove the redirection statements in .htaccess, everything works fine again. I will have a look tomorrow.

m0urs, 2023-06-18:
My pull request was wrong. I recheck tomorrow.

closed 2023-06-20 by magicbug