magicbug/Cloudlog #2230: None of your Logbooks are configured to export data to the QO-100 Dx Club's API.

do4bmw, 2023-06-29:
HY

i want to Upload my Logs die QO100DX Club , and it works in the past
but now i get this Message in Red 

> None of your Logbooks are configured to export data to the QO-100 Dx Club's API.
> To configure this feature, go to your profile page at the [QO-100 Dx Club](https://qo100dx.club/) and Create an API key. Then, navigate to your [Station Locations](http://log.do4bmw.de/index.php/station) and configure you station with the key you have created at the club.

Problem is , i have a Valid API Key allready entered

[image]

phl0, 2023-06-29:
I think the error message is misleading. It also appears if all QSOs are uploaded and no new ones exist. Maybe this is the case here?

do4bmw, 2023-06-29:
Hy Phl0

you are right ... many thanks

[image]

closed 2023-06-29 by do4bmw

phl0, 2023-06-29:
Proposed a fix in https://github.com/magicbug/Cloudlog/pull/2231.