magicbug/Cloudlog #2238: eQSL- / LotW-Import doesn't care about station_id

int2001, 2023-07-02:
**To Reproduce**

Prequisites:
* Working and logging with special Event Call (let's call it "DL0TESTING"). Station-Profile is set up to DL0TESTING
* You are doing a qso with another station (let's call it "DL0FOO") and logging it.
* DL0FOO asks you to log also your personal call. You accept and log it in another Cloudlogaccount also for your personal Call (let's call it DL1BAR)
* Now two qso exists. One DL0TESTING <-> DL0FOO, and one DL1BAR <-> DL0FOO. Same time_off, same grid, etc. - distinguishable via station_id

The Bug:
* You're getting an eQSL (and/or a LotW-Confirmation - that's the second part of the Bug) but only for the QSO between DL1BAR and DL0FOO.
* Cloudlog imports this one and doesn't take care of the station_id. So it updates BOTH QSOs in the Database. In other Words: Even if the Account/Station_id DL0TESTING doesn't have any eQSL or LotW-Credentials, the QSO will also be updated for DL0TESTING. This happens with eQSL and LotW

int2001, 2023-07-03:
Need help, perhaps from @AndreasK79 (?) or @phl0 (?), with the eQSL-Part.

eQSLs-confirmations aren't coming with a gridsquare. So one needs to iterate through the (valid) station_id of the origin-call given for the confirmation. The following functions need to be adjusted for that:
- find_correct_station_id (actually returns only one row) https://github.com/magicbug/Cloudlog/blob/06656bfef81fa9234f2e296950e48f514ec1dec1/application/models/Logbook_model.php#L3773
- of course import itself: https://github.com/magicbug/Cloudlog/blob/06656bfef81fa9234f2e296950e48f514ec1dec1/application/libraries/EqslImporter.php#L134
- import_check: https://github.com/magicbug/Cloudlog/blob/06656bfef81fa9234f2e296950e48f514ec1dec1/application/models/Logbook_model.php#L2472
- eqsl_dupe_check: https://github.com/magicbug/Cloudlog/blob/06656bfef81fa9234f2e296950e48f514ec1dec1/application/models/Eqslmethods_model.php#L176
- eqsl_update: https://github.com/magicbug/Cloudlog/blob/06656bfef81fa9234f2e296950e48f514ec1dec1/application/models/Eqslmethods_model.php#L159

Idea is to update every QSO, which belongs to station_ids (plural) where the ownercall is listed. 
I see no other way, to do that. Problem is - from my sight - minimal, because one would not log a qso with the same origincall from more than one location. so gridlocator could be omitted. should be even better to update every qso (regardless of the origin-call) which matches time, band and operator.

If there is more than one user on the same instance there's a high risk that user "b" also works the same station within the same band/time like user "a" did (rare-stations, etc.). it's far away from hypothetic.

Any more ideas on that?

int2001, 2023-07-03:
Second approach, inspired by https://github.com/magicbug/Cloudlog/issues/2189:
Use COL_STATION_CALLSIGN instead of station_id-Voodoo.

Works for `Import directly from eQSL` but breaks on `Import from file...`

AndreasK79, 2023-07-03:
@int2001 I'll see if I can come up with some suggestions here, but seems I need to dig into the code before I hopefully can offer anything.

int2001, 2023-07-04:
That's really tricky.
I thought implementing a dropdown with StationLocations at the "import eQSL-ADI"-Form could be a solution.
But if one is using the same Call with different locations and the same eQSL-Nick (not unusual when roving or doing pota), it's not userfriendly. 
Problem exists because there's no reference for the station_callsign in the eQSL-ADIF.
Updating every known matching QSO which belongs to the users station_id could be problematic.

Problem is: If one is roving through the grids the eqsl_update is able to update more than one qso for only one eQSL-Confirmation. Ok, that's much better than the current solution where EVERY QSO within band/time_range would be updated (also those which don't belong to the logged in User). But i'm not happy with that.

Very tricky

commit ca850adf (2023-07-05): Fixed #2238
commit da9e6401 (2023-07-05): Merge pull request #2249 from int2001/dev

Fixes #2238