magicbug/Cloudlog #2244: Docker container doesn't setup permissions correctly

alexjj, 2023-07-03:
I installed from the latest image on https://hub.docker.com/r/2m0sql/cloudlog/tags but was presented with a permissions error on the /install/ page, suggesting to `chmod 777`.

I exec'd into the container and ran the write permissions as per this wiki [page](https://github.com/magicbug/Cloudlog/wiki/Installation), which resolved the issue.

Be nice not to have to do that :) Thanks.

magicbug, 2023-07-04:
Yep seems a good thing todo @iu2frl whats the best way to adjust the image template.

iu2frl, 2023-07-04:
@alexjj can you please tell me which steps you did? i tried both arm64 and amd64 and seems to work fine on my VMs

alexjj, 2023-07-04:
I made a compose file with the following:

```
  cloudlog:
    image: 2m0sql/cloudlog:latest
    container_name: cloudlog
    ports: 
      - 8096:80
    volumes:
      - /home/alex/cloudlog/uploads:/var/www/html/application/uploads
      - /home/alex/cloudlog/backup:/var/www/html/application/backup
      - /home/alex/cloudlog/config:/var/www/html/application/config
     restart: unless-stopped
```

`sudo docker-compose up -d`

Then went to ip:8096 and it said words to the effect of "permission issue please do `chmod -R 777 /var/www/html/application`

As the docker container runs as root I thought it might be permissions on the host side, but changing owner on the host machine didn't help. So I then ran the write permissions within the container and that fixed it.

iu2frl, 2023-07-04:
Thanks, i'm testing it

alexjj, 2023-07-04:
If you've not had issues, then I suspect it's to do with the permissions of those volumes and the interaction between host and container. I'm just an application operator so don't know how to change things unless the container image provides an environment option to change the user/group.

iu2frl, 2023-07-04:
@alexjj can you try this setup and let me know if it works:

```yaml
version: '3'
services:
  cloudlog-mysql:
    image: mysql:latest
    container_name: cloudlog-mysql
    environment:
      MYSQL_ROOT_PASSWORD: password-segreta # <- Use a strong password here
    #ports:
      #- "3306:3306" # <- Uncomment these lines only if you need to expose the MySQL container to the network
#    volumes:
#      - cloudlog-dbdata:/var/lib/mysql
    restart: unless-stopped

  phpmyadmin:
    image: phpmyadmin:latest
    container_name: cloudlog-phpmyadmin
    depends_on:
      - cloudlog-mysql
    environment:
      PMA_HOST: cloudlog-mysql
      PMA_PORT: 3306
      PMA_ARBITRARY: 1
      PMA_USER: root
      PMA_PASSWORD: password-segreta # <- Put same strong password here
    restart: unless-stopped
    ports:
      - "8083:80"
      
  cloudlog:
    image: 2m0sql/cloudlog:latest
    container_name: cloudlog-main
    depends_on:
      - cloudlog-mysql
#    volumes:
#      - cloudlog-config:/var/www/html/application/config
#      - cloudlog-backup:/var/www/html/application/backup
#      - cloudlog-uploads:/var/www/html/application/uploads
    ports:
      - "8086:80"
    restart: unless-stopped

#volumes:
#  cloudlog-dbdata:
#  cloudlog-config:
#  cloudlog-backup:
#  cloudlog-uploads:
```

Nothing is persistent here, it is just for testing. Also, which permission did you set? All the steps in the page you linked are already included in the image creation

closed 2023-07-04 by magicbug

iu2frl, 2023-07-05:
@alexjj please follow this guide: https://github.com/magicbug/Cloudlog/wiki/Executing-the-full-standalone-Cloudlog-stack

commit ee90929a (2023-07-04): Merge pull request #2248 from iu2frl/dev

Fixed #2244
commit d225acad (2023-07-04): Merge pull request #2248 from iu2frl/dev

Fixed #2244