magicbug/Cloudlog #2261: FEATURE REQUEST: Awards -> DXCC: Add option to show only unconfirmed QSOs

m0urs, 2023-07-07:
I would like to be able to see a list of all DXCC countries which I have worked but which are not yet confirmed. I would like to use this view to check if I have a possibility to get a confirmation e.g. by requesting a direct QSL card etc.

Currently I do not see a possibility to do this with the current filter options in Awards -> DXCC:

[image]

Maybe it would make sense, to add the same logic to the other Awards view likes WAS etc.

m0urs, 2023-07-07:
> Uh oh! @m0urs, the image you shared is missing helpful alt text. Check your issue body.
>
> Alt text is an invisible description that helps screen readers describe images to blind or low-vision users. [...]

Is this something which we might be able to switch off? :-)

magicbug, 2023-07-07:
> Is this something which we might be able to switch off? :-)

Nope its a good idea to add ALT text to images, which helps those with visual impairments.

`![Alt text](image URL)`

int2001, 2023-07-08:
I cannot reproduce this @m0urs 
if I set the filters exactly as in your screenshot, I get a list (or map) of all those DXCCs for which I never received a QSL or LotW confirmation.
If there is only one LotW/QSL confirmation from that DXCC, the DXCC is not shown, otherwise it is listed.
Isn't that what you wanted?

Maybe I misunderstood your feature-request?

m0urs, 2023-07-08:
@int2001 You are absolutely right. I have no idea what I have seen and/or thought yesterday :-( It is exactly how it should be ... So sorry for that and we can close it.

closed 2023-07-08 by m0urs