magicbug/Cloudlog #2268: Feature Request: Custom Paper Types for Labels

N9EAT, 2023-07-08:
Would like the ability to specify different paper types beyond just Letter and A4 for the Label printing tool. For example, I use a QL-550 Label printer which can accommodate several different label roll sizes. I currently use 2-3/7" width labels and generally print with a height of 1-1/7".

gx1400, 2023-07-20:
+1 Vote request

I recently bought a Brother QL800 series label printer specifically for printing individual shipping labels and QSLs. Would love to print straight from Cloudlog

AndreasK79, 2023-08-03:
It's coming #2367 was just added.

AndreasK79, 2023-08-03:
And now merged into dev, so please test.

N9EAT, 2023-08-04:
> And now merged into dev, so please test.

Works great, have a few comments.

- Paper type only gave me the millimeters option. 

- Might be good for the label to indicate date as being in DD.MM.YY to avoid confusing Americans.  Time UTC or Z would also be nice but not as necessary I believe. 
- option to print grid (or other changing item such as state/county) I was operating from

int2001, 2023-08-04:
Papertype: Yes, we waited for that Feature-request, already. Background: if you can set measurement at Labels **and** Papers, we've to deal with mixed informations. There are many calculations done in the code and everything will be downcalculated to mm (e.g. if inches were set at label). It's tricky, but feasible. I'll take a look at that

Date/Time: First version was with fulldate and UTC in Header. space is very limited on labels. We're currently only able to print text and those ascii-tables. See https://github.com/magicbug/Cloudlog/discussions/2164#discussioncomment-6575812

Grid-/Maidenhead Locator: As space on labels is very limited: Do you have an idea where to put that?

magicbug, 2023-08-04:
Yep i like that!

N9EAT, 2023-08-04:
> Papertype: Yes, we waited for that Feature-request, already. Background: if you can set measurement at Labels **and** Papers, we've to deal with mixed informations. There are many calculations done in the code and everything will be downcalculated to mm (e.g. if inches were set at label). It's tricky, but feasible. I'll take a look at that
> 
> Date/Time: First version was with fulldate and UTC in Header. space is very limited on labels. We're currently only able to print text and those ascii-tables. See [#2164 (reply in thread)](https://github.com/magicbug/Cloudlog/discussions/2164#discussioncomment-6575812)
> 
> So, if @magicbug confirms, i can change the header to "DD.MM.YY UTC": [image]
> 
> Grid-/Maidenhead Locator: As space on labels is very limited: Do you have an idea where to put that?
> 
> Update. Did a little research regarding the Date. Here is a physical QSL-Card which i received from an US-Ham. They also use the DD.MM.YY-Format in our hobby.
> 
> [image]

This looks great. I took a look at my qsl cards and for US hams most who used DD MM YY format had it stated as such on the card. Have plenty who use MM DD YY or even YYYY MM DD format as well - since the space is there I feel clarifying what format is good as I imagine most cases of actually sending a qsl card nowadays are for those hams who do not use LotW / etc. online qsling. I know at least for me if it didn't state as such, I personally would assume a date such as 10/7/22 would mean October 7th, not July 10th. 

Thinking on it some more, with the point of space being limited, it is probably best not to force grid square onto it too - can always manually write change on the card.

int2001, 2023-08-05:
Grids are now available, too.
I think we can close this issue now?!

magicbug, 2023-08-05:
Yep think its safe to close this as complete woop!

closed 2023-08-05 by magicbug