magicbug/Cloudlog #2290: Bug:

rtoenniges, 2023-07-14:
**Describe the bug**
By going to the eQSL page in the QSO menu the error:
```
A PHP Error was encountered
Severity: Warning

Message: Undefined variable $i

Filename: view_log/qso.php

Line Number: 565
```
showes up.
After upgrade vom 2.4.4 to 2.4.5 this error occurs.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Dashboard
2. Click on a QSO that have a received eQSL card
3. Select eQSL
4. See error

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Chrome
 - 11.4.0

phl0, 2023-07-14:
Cannot reproduce the issue here. Can you please tell which operating system, PHP version and mySQL you are using?

rtoenniges, 2023-07-14:
> Cannot reproduce the issue here. Can you please tell which operating system, PHP version and mySQL you are using?

Hi, my server has the following specs:

OS: AlmaLinux 8.8
PHP: 8.0.27
mySQL:  MariaDB 10.3.35

phl0, 2023-07-14:
Never mind. Error found. Fixed in https://github.com/magicbug/Cloudlog/pull/2291.

rtoenniges, 2023-07-14:
Thank you very much!
I can confirm this works.

closed 2023-07-14 by rtoenniges