magicbug/Cloudlog #2294: [Feature Request] Cloudlog as Progressive Web App

HB9HIL, 2023-07-15:
It would be awesome to make Cloudlog able to work as a Progressive Web App

This would allow to run it "offline" on Desktop and Mobile Devices, it feels like an native App and it would improve the UX on mobile devices.

What do you guys think about that.

HB9HIL, 2023-07-15:
Was already discussed a year ago

https://github.com/magicbug/Cloudlog/discussions/1482

dg9vh, 2023-07-15:
We have a project called CloudlogOffline that brings offline logging capabilities and some more to mobile devices. But I also would favorite a PWA of Cloudlog with most features like statistics, log manipulations and so on. But I can imagine most actions would be impossible because this means we have to create a synchronization between the PWA and the database and we also have to keep (for example last 1000) QSOs in an offline-mirror-database within this app.

But in general this could be cool.

HB9HIL, 2023-11-14:
Is now implemented.

HB9HIL, 2023-11-14:
PS: online only

closed 2023-11-14 by HB9HIL