magicbug/Cloudlog #2305: Bug: Previous imported eQSL matches shown as duplicated

rtoenniges, 2023-07-17:
**Describe the bug**
After import QSO Matches directly from eQSL it will show the same QSO every next import as "found duplicated".
As a workaround I have to toggle the eQSL received in the log entry. 
This workaround only works If I toggle the eQSL state minimum one day after I made the QSO.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to "eQSL Import/Export"
2. Click on "Import eQSL QSO Matches"
3. See newly imported eQSLs
4. Refresh Cloudlog
5. Click on "Import eQSL QSO Matches" again
6. Last imported QSOs will show as "found duplicated" from now on

**Expected behaviour**
After successfull import the QSO should not be displayed every time as duplicated.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Chrome
 - Version: 114

Cloudlog 2.4.5

int2001, 2023-07-18:
Everytime you press "Download"/"Import" at the eQSL-Tab, Cloudlog checks for the last received QSL from eQSL in your Logbook.
This is done by date, because eQSL only accepts a date as filter.

So, if you're doing the Download/Import all day long, you will get "dupe QSLs" from eQSLs. Cloudlog ignores these dupes, because the eQSL is already imported.

On the next day the game begins new. The only thing, that would be considerable, is to supress those "duplicate"-Messages to the User. But in fact eQSL delivers the same eQSLs again... 

Your QSOs are marked properly and the eQSL is downloaded once, or did i miss something?

rtoenniges, 2023-07-18:
You're right. The QSO marked correctly and the eQSL is downloaded only once.
But the duplicated message will be displayed every time (days, weeks after) I import new eSQLs until I manually reset/set the eQSL received status. 
If i do not do this, there may be hundrets or more eQSL that show as duplicatedt every time I import new eQSLs.

This does not happen if the eQSL will be received minimum one day after I log the QSO.
In this case the eQSL will be imported and on next import dont show a duplicated.

I dont now if this is a bug or simply the normal workflow...

rtoenniges, 2023-07-18:
> So, you get the same QSOs delivered from eQSL one / two days after? Let me understand by an example:
> 
> 10.07.2023: Doing a QSO with me (DJ7NT) and logging it. 
> 10.07.2023: I send u an eQSL 
> 10.07.2023: you press Import. eQSL will be imported. 
> 10.07.2023: you press import again. eQSL will be shown as "dupe" (which is technically correct, no need to worry here)
> 12.07.2023: you are doing some other QSOs, but not working me again 12.07.2023: you press import. Now the eQSL with me (10.07.2023) is listed as dupe again??

Yes, this is the behavior I encountered.
I will try to reproduce this in the next 3 days and write detailed step-by-step and make more screenshots.

rtoenniges, 2023-07-25:
After a little bit of testing with different states, the behavior seems to be ok.

The last imported eQSL show as long as no new eQSL was received.
If a new eQSL was received the next day or later it will show the old and the new eQSL.
After import again, the old eQSL is gone.

So now very very old eQSL will be shown.

Sorry for the wrong report... I'll close the issue

closed 2023-07-25 by rtoenniges