magicbug/Cloudlog #2308: Columns mismatch after upgrading to 2.4.5

iu2frl, 2023-07-18:
**Describe the bug**
When upgrading from latest release to 2.4.5 some DB structures have been changed, after the upgrade i see:

- Main page:
[image]

- User settings:
[image]

- I tried updating my user settings and:
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu server
 - Browser: Chrome

magicbug, 2023-07-18:
Looks like the migration system didn't successfully complete you could try index.php/migration and see if anything errors.

iu2frl, 2023-07-18:
I think i got it, i had to manually open the DB and create this structure:

[image]

iu2frl, 2023-07-18:
I just pulled the container and executed that, then i opened the usual URL, it asked me the login details and the main page showed up with that error

closed 2023-07-19 by magicbug

iu2frl, 2023-08-18:
@magicbug i just discovered some other columns are missing somewhere else, i tried the migration URL but no luck, any other suggestion?

[image]