magicbug/Cloudlog #2316: Unable to save location

dexdeadly, 2023-07-24:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
On a fresh install I am having an issue. When I try to create an active station I was unable to. I found that I was able to save only a few top items and get passed. However I noticed if I try to save any location information such as the city, state, country and county. I've tried to reboot the docker to no avail. 

**To Reproduce**
Steps to reproduce the behaviour:
1. New Installation
2. Click on user name up top
3. click on Station Locations
4. Try creating the location

**Expected behaviour**
All data to be saved

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Brave and Chrome
 - Version 1.56.11

**Smartphone (please complete the following information):**
None

**Additional context**
Add any other context about the problem here.

int2001, 2023-07-24:
did you create a Station Logbook 1st?
Process is:
* create user
* create Station Logbook
* create Station Location
* Link both

magicbug, 2023-07-24:
Can we first check that in Station Location when your trying to create one that you can see the DXCC dropdown? if not you'll need to fix that first, you can update the tables by going to Admin -> Update Country files

dexdeadly, 2023-07-24:
AH HA. Ok, so I did the logbook, location, linked them and all of that.  The part I did not do was what magicbug had just mentioned. I updated the country files. Once I did that it was able to save everything.

closed 2023-07-25 by dexdeadly