magicbug/Cloudlog #2318: Sat name in ADIF for TEVEL sats

ok1cdj, 2023-07-24:
Hello,  i have problem when uploading adif to Clublog. The TEVELs are not recognized by Clublog.
In Cloudlog is in ADIF TEVEL-X with dash, but Clublog expecting name without dash according to LoTW specs:
` <satellite name="TEVEL1" startDate="2022-01-13">Tevel-1</satellite>`
I think that LoTW accept both, because i have confirmed QSOs. Problem is only with Clublog.

Have a nice day.

73 Ondra OK1CDJ

magicbug, 2023-07-24:
LoTW reduces the name due to constraints in its system design, but I don't get why Clublog wants to change the name of the satellite.

ok1cdj, 2023-07-24:
I created issue in Clublog:
https://clublog.freshdesk.com/support/tickets/74918

I don't like change the adif manualy, using upload by cron.

magicbug, 2023-07-24:
Worse case can build in a remap like we use for LoTW but would prefer not to

ok1cdj, 2023-07-24:
Or persuade G7VJR to make change on Clublog side.

closed 2023-08-30 by magicbug