magicbug/Cloudlog #2334: Uploading .p12 LoTW certificate fails with no visible error

lu1aax, 2023-07-27:
Using latest version updated with git, using it in developer mode now.

I needed to update my LoTW. I uploaded new p12 file. No error, but new cert not shown. I deleted the old cert. Tried again and no error and no available certs listed.

Please help, thanks!

phl0, 2023-07-27:
This is most probably not a Cloudlog issue but OpenSSL. See https://github.com/magicbug/Cloudlog/issues/2068.

lu1aax, 2023-07-27:
Hi! Thanks for your prompt reply.

Activated the legacy support in openssl.cnf. Still the same situation.

There's no trace of what went wrong... No information in the logs, it's strange. Even though I am in developer mode there's absolutely no feedback about what went on.

lu1aax, 2023-07-28:
It seems like there is something else going on here, I cannot upload the old certificate (2020) neither.

phl0, 2023-07-28:
Did you edit the openssl.cnf file which is in use for the PHP module running inside the webserver? 
It does not matter if yoj are uploading old or new certs since all of them use the same (old) algorithms as to my knowledge. Set the log threshold to 4 and have a look at the logs generated in application/logs.

lu1aax, 2023-07-28:
Thanks for the guidance!

I'm getting this:

ERROR - 2023-07-28 19:14:41 --> Severity: Notice --> Undefined index: cert /srv/www/lu1aax.taslik.net/Cloudlog/application/controllers/Lotw.php 407
ERROR - 2023-07-28 19:14:41 --> error:0308010C:digital envelope routines::unsupported

closed 2023-07-28 by lu1aax