magicbug/Cloudlog #2341: ADIF Error

DL3EL, 2023-07-28:
I have just upgraded to the latest version and now I see an error while importing ADIF data. The same files worked without issues before.
Interesting is, that the error is empty (see photo). I will attach the file for import as well. [
[image]
](url)
[cl_error.txt](https://github.com/magicbug/Cloudlog/files/12196826/cl_error.txt)

vy 73 de Thomas, DL3EL

DL3EL, 2023-07-28:
additional remark: when I export a single qso from the advanced view, change call and time and import it again, the same error occures

int2001, 2023-07-28:
the main question is: Are all QSOs imported? Do you miss a QSO?

There's bug in the last master, which is fixed here: https://github.com/magicbug/Cloudlog/commit/49dcc9c3d0893032531910c500bbdf109ec724b0
will be available in the next Version.

The bug consists of printing out that Error, but importing everything...

phl0, 2023-07-28:
Would you be able to supply the ADIF file you were trying for debugging purposes? If so you may send it to me via email pse.

DL3EL, 2023-07-28:
it looks like everything is imported ok

int2001, 2023-07-28:
Ok, then this is fixed in the latest dev (and soon in master).

But can be closed?!

DL3EL, 2023-07-28:
I think so

closed 2023-07-28 by DL3EL