magicbug/Cloudlog #2342: A locator longer than six characters or faulty info

oh1jw, 2023-07-28:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
A locator longer than six characters, usually fetched from QRZ.com or similar, gives warning
when saving qso - but won't save qso details or actually the qso at all. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'QSO (live or post)'
2. Add qso details, with long grid e.g. IN52OF72PK
3. Click on 'Add qso'
4. See error --> 
[image]

**Expected behaviour**
Shorten locator value automaticly for 6 characters? OR at least save qso without locator, but not losing all the qso details and give some kind of warning about missing locator.

**Screenshots**
[image]

magicbug, 2023-07-28:
Not so much faulty just not accepted, works in progress to handle them however.

magicbug, 2023-07-28:
The dev branch can handle up to 8 there for I'm going to propose we trim anything beyond that going forward.

oh1jw, 2023-07-29:
> Not so much faulty just not accepted, works in progress to handle them however.

Good that in progress... this happened me in the middle of the pileup, so there was already a lot of hassle when I noticed that the entire qso was not saved and the details were cleared.

magicbug, 2023-07-30:
Now resolved in the dev branch

closed 2023-07-30 by magicbug