magicbug/Cloudlog #2343: Adding qso doesn't look right

DL3EL, 2023-07-28:
I have just upgraded to the latest version , afterthat I have rebootet my system. When I enter qso via Live or Post QSO several things seem odd:
- there is not update from qrz.com any more
- the suggestion window is always empty
- after pressing save, it takes ages until the qso is accepted
- if it is accepted, the DXCC is always empty

When I add via Live Contest Logging, all of the above works, except the qrz.com update

Did I miss something during the update? I did it via the update script.

vy 73 de Thomas, DL3EL

magicbug, 2023-07-28:
Have you made sure you have callbook data setup in /application/config/config.php as per the documentation?

DL3EL, 2023-07-28:
yes, it is there, I did not change it and it worked until the update

magicbug, 2023-07-28:
Depending where you are updating from can indicate different things, I'd suggest turning on logging in  /application/config/config.php and then check /application/logs after trying to log a QSO to see what errors are chucked out. Might be wise to put Cloudlog into development mode too.

DL3EL, 2023-07-28:
I have switched logging to debug mode, see attached output. Development mode gives no info on the add qso masks
[log-2023-07-28.php.txt](https://github.com/magicbug/Cloudlog/files/12198701/log-2023-07-28.php.txt)

phl0, 2023-07-29:
I looks like the callsign to lookup is undefined. Are you using the latest release aka v2.4.5? or the dev branch? Could you switch on the dev tools of your browser top see of there are any (JS) Errors?

DL3EL, 2023-07-29:
Success!! I used the js logging of my browser and found:
[image]
Then I looked into the code and found out, that there is one change in application/controllers/Lookup.php, in the function scp(). 
I reverted the change and everything is back to normal :-)

And I did the update yesterday evening with the then available software on the master branch. In the config sample it says 2.4.5.

phl0, 2023-07-29:
Could you please point out which change you reverted?

DL3EL, 2023-07-29:
I compared the actual version against my last, from dec 2022 (last update, lazy me). There have been two changes, but I think this one did it:

Dec 2022:
	public function scp($call) {

		if($call) {
			$uppercase_callsign = strtoupper($call);
		}

		// SCP results from logbook

Jul 2023:
	public function scp() {
		if($_POST['callsign']) {
			$uppercase_callsign = strtoupper($_POST['callsign']);
		}

		// SCP results from logbook

int2001, 2023-07-29:
Have you tried from a different browser/PC - and is the problem still there when using another PC/Browser?

I like to exclude a caching-issue here

DL3EL, 2023-07-29:
I have tried it with my android tablet, same result

int2001, 2023-07-29:
ok, and with your patch and the android-tab (where you are using a fresh browser which has never seen Cloudlog before, or incognito-mode) it is working?
strange...

DL3EL, 2023-07-29:
no, I was not using "a fresh browser which has never seen Cloudlog before, or incognito-mode", just another browser. And in reality it was no "patch", I just rolled back that specific file to the latest version I had, that was from Dec 2022

int2001, 2023-07-29:
yes, i understand.
the strange thing is:
- the scp() method was changed 3 weeks ago from GET to POST
- so the JS at qso-form was also changed to POST

Let me try to explain:
at the 2022-Version the scp at lookup.php is based on GET. 
Normally your Codebase is now incompatible, because JS it GET, and lookup.php is POST (due to your rollback)
The only thing, that would explain this behaviour is caching (old JS is cached).

DL3EL, 2023-07-29:
understand. Is there any chance to enforce a clearing of the cache? I just tried the "patched" version on my desktop with the incognito-mode, worked as well.

int2001, 2023-07-29:
not really,
but you can look into your browser (Dev-Tools - see screenshot). Do you see the same in your CL?

[image]

phl0, 2023-07-29:
> Is there any chance to enforce a clearing of the cache?

For me with Firefox on Ubuntu normally CTRL + F5 does the trick.

DL3EL, 2023-07-29:
with the help of @int2001 we could fix the issue. Problem was, that during after update the file assets/js/sections/qso.js was not refreshed in the browser. After manually doing this, everything is ok now.

closed 2023-07-29 by DL3EL