magicbug/Cloudlog #2346: Advanced Logbook Filter is missing qsos

DL3EL, 2023-07-29:
you can filter qsos, if they are cionfirmed with LOTW or not. I have several qsos, which are not shown, when you select LOTW = NO. 
A closer look at the qso data shows, that these qso do not contain the field "<LOTW_QSL_RCVD:1>N".

Could that be the reason?

phl0, 2023-07-29:
> A closer look at the qso data shows, that these qso do not contain the field "<LOTW_QSL_RCVD:1>N".

What exactly do you mean by "does not contain the field"? Are you talking about an ADIF im- or export?

phl0, 2023-07-29:
Oh advanced logbook. My fault for bot reading correctly. @AndreasK79 already has an idea. So this should be fix-able.

phl0, 2023-07-29:
I proposed a patch with https://github.com/magicbug/Cloudlog/pull/2347. @DL3EL could you please verify this works on your system?

AndreasK79, 2023-07-29:
It's merged now, so pull latest dev.

DL3EL, 2023-07-30:
@AndreasK79 with that it works, thanks

closed 2023-07-30 by DL3EL