magicbug/Cloudlog #2364: Undefined index: station_callsign

Bastiti, 2023-08-02:
Hello,

Installed a fresh docker container with cloudlog

After a nice install, I got this error when I try to upload a 300entry ADIF:

A PHP Error was encountered
Severity: Notice

Message: Undefined index: station_callsign

Filename: models/Logbook_model.php

Line Number: 2752

Do you have any idea on how it can be fixed ?

Best regards,
ON4BCY

int2001, 2023-08-02:
Can you post an example row/dataset of your adif, which you're trying to import?

Smells a little like your adif doesn't contain your own Callsign.

Bastiti, 2023-08-02:
here is my adif
[ON4BCY.txt](https://github.com/magicbug/Cloudlog/files/12243975/ON4BCY.txt)

int2001, 2023-08-02:
Imported the whole adif with the latest version successfully here. Can not reproduce the bug.

Only one QSO was dupe: Date/Time: 2022-10-10 07:16:00 Callsign: EB4DOL Band: 15m Duplicate

Please give us some information about your stack. Versions, webserver, etc. a good summary can be found in Cloudlog under "Admin -> Debug Information"

Bastiti, 2023-08-02:
Found out I was able to import by add:
<STATION_CALLSIGN:6>ON4BCY

At the end of lines, then it works.

Bastiti, 2023-08-02:
[image]

magicbug, 2023-08-03:
If this is happening on the docker image I think it's because it didn't pick up the bug fix, I can report this is resolved in the code.

Aiming to get a new stable release out this week.

magicbug, 2023-08-03:
It might resolve using the latest image rather than the tagged

Bastiti, 2023-08-03:
Thanks,

I'll try this, hope it'll resolve as leaving soon for YOTA Camp