magicbug/Cloudlog #2365: API issue with QSO endpoint

OuiWii, 2023-08-02:
When trying to access the /api/qso endpoint using GridTracker, it displays "Invalid Response". I tried to request myself from the server and I got "station id does not belong to the API key owner.". I have thoroughly checked via SQL, front-end and API that the station exists and is connected to user id 6, same as the API key.

Steps to reproduce the behaviour:
1. Use your program of choice to connect to CloudLog QSO API. Leave HTTPS off if using Wireshark.
2. Error message will appear in server response - "station id does not belong to the API key owner."

**Expected behaviour**
I would get a 200 OK response with data confirming that the ADIF data has been read.

 - OS: Win10 x64
 - Browser: Not applicable
 - Version: Not applicable

 - Server OS: Raspbian Jammy ARM32
 - Stack: LAMP

int2001, 2023-08-02:
Gridtracker supports CloudLogs Station-ID since Version 1.22.0725 (september last year, i think).

Did you fill in one of your station_ids (as seen in settings -> station locations) into GridTracker?

That field in Gridtracker-Settings must be a station_id which corresponds with one of your locations (see screenshot)

[image]

OuiWii, 2023-08-02:
Yep, I have. Both station IDs return same messages. Even made a third one
to test.

magicbug, 2023-08-03:
I'd suggest trying cloudlog aurora to see if that works https://aurora.cloudlog.org/

OuiWii, 2023-08-03:
I’ve tried it but it doesn’t give much feedback on if it’s working at all.
I will attempt to wireshark it tomorrow.

phl0, 2023-08-03:
> Would you wish me to post the JSON packet for review?

For sure! :)

int2001, 2023-08-03:
Before we dive into that (i believe that the json is correct) which precise Version of CL are you running? 
And which stack (Docker? Plain LAMP within Linux (but without docker)?)

Please share the last commit-id if possible. Shell: go to Cloudlog-Root and type `git log -1`

OuiWii, 2023-08-03:
> Which precise Version of CL are you running?
8e28c1
> And which stack?
LAMP without Docker, using Apache vHosting

> > Would you wish me to post the JSON packet for review?
> 
> For sure! :)
`POST /index.php/api/qso HTTP/1.1
Content-Type: application/json
Content-Length: 56
User-Agent: GridTracker/1.23.0402
x-user-agent: GridTracker/1.23.0402
Host: ***
Connection: close

{"key":"cl**********","type":"adif","string":"<eor>"}`

OuiWii, 2023-08-03:
As a comparison, a real log packet looks like this

```
POST /index.php/api/qso HTTP/1.1
Content-Type: application/json
Content-Length: 348
User-Agent: GridTracker/1.23.0402
x-user-agent: GridTracker/1.23.0402
Host: ***
Connection: close

{"key":"cl**********","station_profile_id":"1","type":"adif","string":"<CALL:6>9A1CCB <GRIDSQUARE:4>JN85 <MODE:3>FT8 <RST_SENT:0> <RST_RCVD:0> <QSO_DATE:8>20230803 <TIME_ON:6>164047 <QSO_DATE_OFF:8>20230803 <TIME_OFF:6>164047 <BAND:3>20m <FREQ:9>14.075399 <STATION_CALLSIGN:6>SA0WII <MY_GRIDSQUARE:6>JO99DL <DXCC:3>497 <COUNTRY:7>Croatia <EOR>"}```

Note that in the test packet, no profile ID is submitted. I will submit this to GridTracker for further review. Thank you for all!

phl0, 2023-08-03:
> Note that in the test packet, no profile ID is submitted. I will submit this to GridTracker for further review. Thank you for all!

Yeah. And this is useless with V2. Also why would it send an empty ADIF record? It would be better to use the API-key test functions. Results in an info if key is rw or not.

phl0, 2023-08-03:
> I would presume this is because you have to enter the full API url, https://yourserv.er/index.php/api/qso, and not the base URI.

Hmm yeah. But still it would make sense to change that and analyze the result of the test API key function.

OuiWii, 2023-08-03:
> Hmm yeah. But still it would make sense to change that and analyze the result of the test API key function.

It probably would. I'll leave that to someone else though, I have to get back to translating CloudLog to Swedish :)
See merge request https://gitlab.com/gridtracker.org/gridtracker/-/merge_requests/279 on GitLab for "continuation"