magicbug/Cloudlog #2381: New Docker installation issues.

dexdeadly, 2023-08-07:
I'm currently setting up a new install of cloudlog. I have portainer running and I have a mariadb server with a cloudlog table and user to use. Here is my compose for cloudlog.

`version: '3.7'
services:

cloudlog:
image: 2m0sql/cloudlog:latest
container_name: cloudlog
restart: unless-stopped
volumes:
- /dockers/cloudlog/config:/var/www/html/application/config
- /dockers/cloudlog/backup:/var/www/html/application/backup
- /dockers/cloudlog/uploads:/var/www/html/application/uploads

networks:
default:
external:
name: dockers_default`

When I start the docker it loads and creates these 3 directories with no problems. I then go to the url I have setup in NPM and it loads and says the following

`Install Cloudlog
Please make the /application/config/ folder writable. Example:

chmod -R 777 /application/config/

Don't forget to restore the permissions afterwards.`

I noticed the folders created when I viewed them were owned by root:root. I go ahead and run the command for that folder and get the install page. I put in the information and then I get the following issue.

`
Warning: unlink(/var/www/html/install/assets/install.sql): Permission denied in /var/www/html/install/index.php on line 14

Warning: rmdir(/var/www/html/install/assets/): Directory not empty in /var/www/html/install/index.php on line 16

Warning: unlink(/var/www/html/install/config/config.php): Permission denied in /var/www/html/install/index.php on line 14

Warning: unlink(/var/www/html/install/config/database.php): Permission denied in /var/www/html/install/index.php on line 14

Warning: rmdir(/var/www/html/install/config/): Directory not empty in /var/www/html/install/index.php on line 16

Warning: unlink(/var/www/html/install/includes/core_class.php): Permission denied in /var/www/html/install/index.php on line 14

Warning: unlink(/var/www/html/install/includes/database_class.php): Permission denied in /var/www/html/install/index.php on line 14

Warning: rmdir(/var/www/html/install/includes/): Directory not empty in /var/www/html/install/index.php on line 16

Warning: rmdir(/var/www/html/install/): Directory not empty in /var/www/html/install/index.php on line 16

Warning: rmdir(/var/www/html/install): Directory not empty in /var/www/html/install/index.php on line 16

Warning: Cannot modify header information - headers already sent by (output started at /var/www/html/install/includes/core_class.php:127) in /var/www/html/install/index.php on line 67
Install successful
Please delete the install folder`

I went to the docker and removed the install directory myself. However when I then go to just the URL I configured I am now presented with the following error and then I am stuck.

An Error Was Encountered Unable to determine what should be displayed. A default route has not been specified in the routing file.

I have confirmed that in the config folder I edited the config.php file and added my NPM server there. So that the URL I configured will work. However this doesn't change anything. I'm not sure if this is due to the docker container creating the folders as root or not.

magicbug, 2023-08-07:
@iu2frl any ideas?

iu2frl, 2023-08-07:
I definetly think it is somehow related to bad permissions, can you try with this stack? 

https://github.com/magicbug/Cloudlog/wiki/Executing-the-full-standalone-Cloudlog-stack

iu2frl, 2023-08-08:
so if you open the normal url (without "install") you get such error?

dexdeadly, 2023-08-08:
So if I try and go to just the ip:port it tries to go to /index.php/welcome and /index.php/dashboard

theritvars, 2023-08-15:
Same here. Attempted to install with the default docker stack and got the same PHP errors.
Then deleted the install folder as the page instructed, however, when going to page root it gets stuck in a redirect loop.

[image]

iu2frl, 2023-08-15:
Thanks I will check asap

iu2frl, 2023-08-15:
Seems like i found two problems:
1. The Dockerfile was not setting proper permissions to the `install`folder
2. Empty lines at the end of `install/includes/core_class.php` and `install/includes/database_class.php` were creating the header issue (see [here](https://stackoverflow.com/questions/8028957/how-to-fix-headers-already-sent-error-in-php))

But i still got:
```
Warning: rmdir(/var/www/html/install): No such file or directory in /var/www/html/install/index.php on line 16
Warning: Cannot modify header information - headers already sent by (output started at /var/www/html/install/index.php:16) in /var/www/html/install/index.php on line 67
``` 

The second error is caused by the first one, means that something is writing to the html page before the header function is called at line 67, i think the for loop in `install/index.php` deletes the install folder and then the php page tries to delete it again after the loop, so a workaround was to add this `file_exists` methods before trying to delete an unexisting file:

```php
...
if (file_exists($file)) {
    delDir( $file );
}
...
```

Now there are no more warnings after the installation step but the loop between `dashboard` and `welcome` persists and I don't really think it is caused by the Docker image.

@magicbug any idea? Have you tried to deploy the new version on an empty Linux machine? I tried both arm64 and x86 but i get the same loop on both

andrearota, 2023-08-16:
Same here. I am setting up an instance using 2.4.6 image. I got the same warnings as @iu2frl after the install, I saw already you merged the commits to fix them. Once installation has finished, I get the same infinite redirection welcome -> dashboard -> welcome.

iu2frl, 2023-08-17:
For those who are experiencing issues with redirections I recommend using some older tag for the cloudlog image, the 2.4.4 for example does not suffer the same issue, you can find the full list [here](https://hub.docker.com/r/2m0sql/cloudlog/tags) while we investigate the issue

dexdeadly, 2023-08-17:
if I do 2.4.4 I get 

Unable to determine what should be displayed. A default route has not been specified in the routing file. 

when trying to go to my page

andrearota, 2023-08-17:
Tried. 2.4.4 shows the headers errors, but it works without infinite redirects. Allegedly 2.4.4 running as container suffers of #2162 (I uploaded a test ADIF file and QSO's do not show up, same if I log a new QSO).

iu2frl, 2023-08-18:
@dexdeadly @andrearota just tried with latest release, if at the infinite loop you replace the url with `index.php/user/login` you should be able to login and proceed to the app

riwim, 2023-08-18:
I can confirm the problem with the non-terminating loop also with the variant "Installation on Linux Server" and the release 2.4.7 on a fresh Ubuntu 20.04 LTS system. Likewise, I can confirm that switching to the 2.4.4 release, `git checkout tags/2.4.4`, is a workaround for this issue.

iu2frl, 2023-08-18:
@magicbug i would close this as the permissions have been fixed and track a new anomaly for the infinite loop, what do you think?

closed 2023-08-18 by magicbug

petrkr, 2023-08-26:
I have same problem (infinite loop), so solution is install older version? 

I tried it, but it have problem with deprecated PHP functions.

Any way how to run it on supported environment? (supported means not EOL like php7.4 which ends support this spring)

I  trying use most update secured setup
Debian 12
PHP 8.2

iu2frl, 2023-08-27:
Hi @petrkr after the setup just browse to ipaddress/user/login

petrkr, 2023-08-27:
@iu2frl thanks. So there is some bug with initial redirect from installer. It should point to user/login instead welcome page... But strange is, if I already goes on user/login page, I can not reproduce (after logout) same infinite loop behavior.

iu2frl, 2023-08-27:
Correct, I can't reproduce it either

petrkr, 2023-08-27:
@iu2frl well I found, it is caused by database option version2_trigger.

after first login it is replaced to "true" value. If you put it back to "false" you have loop back again

iu2frl, 2023-08-27:
@magicbug what is that flag used for?

petrkr, 2023-08-27:
well I already know why it happen, but do not know wanted workflow, so do not know how to fix it... 

My guess, since I see this code for like 4 hours since yesterday..

This flag was used in upgrade from 1.x to 2.x version to probably show up something to users, who did upgrade and also probably migrate data.

Because we have new install, we do not have anything like that... But due this flag, there was added logic, where "dashboard" page redirect use to welcome page, in order probably to show some message regarding to upgrade.

But it suppose user is already logged in, because in welcome page is logic, which will redirect back to dash board unless user is logged in... So that is that loop -- dashboard is not happy, because this version flag is still not set, so it redirects back, but welcome is still not happy, because user is still NOT logged in.. so it loops

if you manually go to user/login, you will log in, then you are redirected to dashboard,, which is still not happy from this flag, so it will send you back to welcome page, but that is already satisfied about logged user, so it will do check about this flag,,, and if this flag is not set yet and user had some records, probably something is shown (I not have pre-2.x version, do not know what that would be).... but in our case, no data there, so it just simple set this flag to "True" and that is why, dashboard is satisfied and we can not reproduce this error again

Here is logic, which will send you to welcome page, if flag is false (from dashboard)
https://github.com/magicbug/Cloudlog/blob/master/application/controllers/Dashboard.php#L22

Here is logic, which will send you back to dashboard, if you are not logged in (notice, it is in constructor, so it is BEFORE that index function)
https://github.com/magicbug/Cloudlog/blob/master/application/controllers/Welcome.php#L15

Here is logic, which will do some magic if you do not have set this flag
https://github.com/magicbug/Cloudlog/blob/master/application/controllers/Welcome.php#L19

And here is logic about that flag, which will show something if you have some data, otherwise it will just set that flag
https://github.com/magicbug/Cloudlog/blob/master/application/controllers/Welcome.php#L38

petrkr, 2023-08-27:
@iu2frl so yes,

there is some migration page, which do some db stuff, which probably could not be done by automatic, since they are need be claimed for exact user-id, which could be different across instalations.

[image]

iu2frl, 2023-08-27:
This is upgrade stuff, you're definitely right, it may be used as the upgrade procedure via docker is actually performed by pulling a blank image and connect to an existing DB, I think something in the "check if the database has some data" logic may have broken

iu2frl, 2023-08-28:
@petrkr updated Docker image to PHP 8.2 in https://github.com/magicbug/Cloudlog/pull/2449

hellmind, 2023-09-04:
so isnt working ?

iu2frl, 2023-09-05:
To me, it worked just fine I upgraded my installations yesterday