magicbug/Cloudlog #2402: Set gridsquare in edit QSO not working (anymore)

Werzi2001, 2023-08-13:
**Describe the bug**
I imported QSOs without gridsquare set and want to set the gridsquare now (that's something I basically do every time). Setting the gridsquare is not possible.

**To Reproduce**
Steps to reproduce the behaviour:
1. Create/Import QSO without grirdsquare
2. Open "Edit QSO" dialog
3. Set a gridsqure
4. Correct distance is not set instead the Cloudlog starting page is inserted into the "Edit QSO" dialog
5. Clicking on Save does nothing

**Expected behaviour**
Correct distance being shown and save being possible.

**Additional context**
The services being called are `/index.php/logbook/index/logbook/searchbearing/jn57re/60` and `/index.php/logbook/index/logbook/searchdistance/jn57re/60`. Both do not return the expected stuff but the Cloudlog dashboard (as HTML). Furthermore I get javascript errors `Uncaught Error: Map container is already initialized.`. For me that happens both on `master` and `dev` branches.

magicbug, 2023-08-13:
[image]

Yep can confirm this is a bug

Werzi2001, 2023-08-13:
Thanks for confirmation. Just wanted to point out that it looks differently for me. In my instance the dashboard is added to the dialog (not just "0 km"):

[image]

int2001, 2023-08-13:
Cannot reproduce. See Video (it's the latest master // but on dev also not reproduceable)

https://github.com/magicbug/Cloudlog/assets/1410708/42ccf8ea-a209-4ef2-9485-7b2023fd8171

AndreasK79, 2023-08-13:
Tried on master/dev. Not seeing that problem here.

magicbug, 2023-08-13:
> Cannot reproduce. See Video (it's the latest master // but on dev also not reproduceable)
> 
>  Screen.Recording.2023-08-13.at.18.32.58.mov

Yep saves fine just the recalculated distance doesn't seem to happen for me

int2001, 2023-08-13:
interesting. as seen in vid the distance calculations are ok here on master and on dev.
don't know where to search, since i cannot reproduce that.

magicbug, 2023-08-13:
Now that's weird

Werzi2001, 2023-08-13:
I'm having a hard time to get Cloudlog to work properly. I pulled/fetched everything but still get the "whole dashboard is shown instead of the distance" error. Furthermore I noticed a `Uncaught TypeError: connectButton is null` javascript error. The error is triggered by the winkey.js which seems to be included without condition but the matching connectButton only seems to be created if winkey is enabled.

int2001, 2023-08-13:
Sounds a little like browsercaching issues. Have you tried to wipe your cache?

Werzi2001, 2023-08-13:
I cleared the cache several times but it didn't change anything. Also if I call the `logbook/searchbearing` directly (in a new tab) the same problem occurs (I get the Cloudlog dashboard). I even tried to put a `die()` direct into `Logbook.php#searchbearing` but it also does not change anything. So the function doesn't even get called. Very strange...

PS: I tried the `dev` branch and the winkey.js bug is gone.

int2001, 2023-08-13:
Sorry, but i don't have any more ideas on that. Downgraded my dev to php7.4 - but I still can't reproduce.

The redirect to dashboard happens, when you're not logged in. Last idea is: checking cookies?

Werzi2001, 2023-08-13:
Really strange. As every other request works I don't think the cookies are the problem (and I don't see something odd with them). I did some further testing and noticed that sometimes it works but most of the time (>95%) it does not. Can you tell me where this redirect to the dashboard happens? I want to debug at that point. Thanks a lot.

Werzi2001, 2023-08-13:
I think I might have found it:

- When triggering the "Edit" dialog from the dashboard the request is made to `/logbook/searchbearing/aa00aa/61` (causing an 404 error).
- When triggering the "Edit" dialog from the logbook (Logbook > Overview) the request is made to `/index.php/logbook/searchbearing/aa00aa/61` (causing it to work properly).
- When triggering the "Edit" dialog from the logbook from any page that was chosen from the pagination the request is made to `/index.php/logbook/index/logbook/searchbearing/aa00aa/60` (causing the dashboard to be returned).

Can you confirm that?

I assume that the requests use relative paths and therefore they only work under specific conditions (specific paths).

int2001, 2023-08-13:
Will have a look at that tomorrow. What did you set at "base_url" in config.php?

Werzi2001, 2023-08-13:
I set the full qualified domain name including https://. No hurry. As it works in the Logbook Overview I found a workaround. Thanks a lot for your help.

magicbug, 2023-08-13:
base_url in the javascript wasn't defined which was causing the issue @Werzi2001 if you pull down the latest version should be resolved

Werzi2001, 2023-08-14:
@magicbug I can confirm that it works in the `dev` branch. Thanks a lot for the quick fix.

commit 25aa87f4 (2023-08-13): [QSO] Fixes issue when editing a gridsquare that distance wasn't calculated.

base_url wasn't defined therefore url only worked if you didn't use index.php

This fixes #2402
closed 2023-08-14 by Werzi2001