magicbug/Cloudlog #2416: table adif_modes contains 9672 rows

DL2CC, 2023-08-17:
**Describe the bug**
Table adif_modes contains lots of duplicate modes

**To Reproduce**
Add QSO, open dropdown for mode, see lots of duplicates

**Expected behaviour**
fewer modes with distinct values should be visibke

**Additional context**

Updated to latest version using git pull today, don't know when or by which update this happened as I have updated last few releases without using the QSO entry window. Definitely not happened by update to 2.4.7 (switched to my backup but didn't help) but used to work fine 6 weeks ago when I was using it.

adif_modes contains 9672 rows, 339 DIGITAL VOICE, out of which just C4FM alone has 112 rows

AndreasK79, 2023-08-28:
Looking at the code, I can't tell why this happened. Did you manage to clean it up?

DL2CC, 2023-08-28:
hi there, no not yet. I haven't found the create command for the table yet so I can drop and recreate. Couldn't find it in the setup script. Do you know where it is?

AndreasK79, 2023-08-28:
Haven't looked, but you could just remove duplicates in the table.

AndreasK79, 2023-08-28:
I checked myself. I had a few duplicates, which I removed with the attached SQL. Make a backup, just to be on the safe side.

Try the following SQL:

> DELETE from adif_modes where id in
   (SELECT id from
        (SELECT *, ROW_NUMBER() OVER(PARTITION BY mode, submode, qrgmode ORDER BY  id) as rn
    FROM adif_modes) as delmodes
WHERE rn > 1);

dg9vh, 2023-08-28:
You also could use this query:

`-- phpMyAdmin SQL Dump
-- version 5.2.1
-- https://www.phpmyadmin.net/
--
-- Host: localhost:3306
-- Erstellungszeit: 28. Aug 2023 um 20:17
-- Server-Version: 5.7.41-log-cll-lve
-- PHP-Version: 8.1.16

SET SQL_MODE = "NO_AUTO_VALUE_ON_ZERO";
START TRANSACTION;
SET time_zone = "+00:00";

/*!40101 SET @OLD_CHARACTER_SET_CLIENT=@@CHARACTER_SET_CLIENT */;
/*!40101 SET @OLD_CHARACTER_SET_RESULTS=@@CHARACTER_SET_RESULTS */;
/*!40101 SET @OLD_COLLATION_CONNECTION=@@COLLATION_CONNECTION */;
/*!40101 SET NAMES utf8mb4 */;

--
-- Datenbank: `huebelon_clprod`
--

-- --------------------------------------------------------

--
-- Tabellenstruktur für Tabelle `adif_modes`
--

DROP TABLE IF EXISTS `adif_modes`;
CREATE TABLE `adif_modes` (
  `id` int(9) NOT NULL,
  `mode` varchar(12) NOT NULL,
  `submode` varchar(25) DEFAULT NULL,
  `qrgmode` varchar(4) NOT NULL,
  `active` int(11) NOT NULL
) ENGINE=InnoDB DEFAULT CHARSET=utf8mb4;

--
-- Daten für Tabelle `adif_modes`
--

INSERT INTO `adif_modes` (`id`, `mode`, `submode`, `qrgmode`, `active`) VALUES
(1, 'AM', NULL, 'SSB', 0),
(2, 'ARDOP', NULL, 'DATA', 0),
(3, 'ATV', NULL, 'DATA', 0),
(4, 'C4FM', NULL, 'DATA', 0),
(5, 'CHIP', NULL, 'DATA', 0),
(6, 'CHIP', 'CHIP128', 'DATA', 0),
(7, 'CHIP', 'CHIP64', 'DATA', 0),
(8, 'CLO', NULL, 'DATA', 0),
(9, 'CONTESTI', NULL, 'DATA', 0),
(10, 'CW', NULL, 'CW', 1),
(11, 'CW', 'PCW', 'CW', 0),
(12, 'DIGITALVOICE', NULL, 'DATA', 1),
(13, 'DOMINO', NULL, 'DATA', 0),
(14, 'DOMINO', 'DOMINOEX', 'DATA', 0),
(15, 'DOMINO', 'DOMINOF', 'DATA', 0),
(16, 'DSTAR', NULL, 'DATA', 0),
(17, 'FAX', NULL, 'DATA', 0),
(18, 'FM', NULL, 'SSB', 1),
(19, 'FSK441', NULL, 'DATA', 0),
(20, 'FT8', NULL, 'DATA', 1),
(21, 'HELL', NULL, 'DATA', 0),
(22, 'HELL', 'FMHELL', 'DATA', 0),
(23, 'HELL', 'FSKHELL', 'DATA', 0),
(24, 'HELL', 'HELL80', 'DATA', 0),
(25, 'HELL', 'HFSK', 'DATA', 0),
(26, 'HELL', 'PSKHELL', 'DATA', 0),
[... 208 lines cut ...]

--
-- Indizes der exportierten Tabellen
--

--
-- Indizes für die Tabelle `adif_modes`
--
ALTER TABLE `adif_modes`
  ADD PRIMARY KEY (`id`);

--
-- AUTO_INCREMENT für exportierte Tabellen
--

--
-- AUTO_INCREMENT für Tabelle `adif_modes`
--
ALTER TABLE `adif_modes`
  MODIFY `id` int(9) NOT NULL AUTO_INCREMENT, AUTO_INCREMENT=241;
COMMIT;

/*!40101 SET CHARACTER_SET_CLIENT=@OLD_CHARACTER_SET_CLIENT */;
/*!40101 SET CHARACTER_SET_RESULTS=@OLD_CHARACTER_SET_RESULTS */;
/*!40101 SET COLLATION_CONNECTION=@OLD_COLLATION_CONNECTION */;`

closed 2023-08-30 by magicbug

DL2CC, 2023-10-09:
Thanks a lot dg9vh & AndreasK79! 
dg9vh sql worked like a charm. Now I finally can use live QSO again.

dg9vh, 2023-10-09:
Nice to hear :-)