magicbug/Cloudlog #2424: Message: Call to undefined function lang() on user/login.php

StrangeWill, 2023-08-23:
Upon upgrading to the latest version (v2.4.7), both my docker image (from before an official Docker image was available) and the image provided via `2m0sql/cloudlog` do not work with the following error:

```
An uncaught Exception was encountered
Type: Error

Message: Call to undefined function lang()

Filename: /var/www/html/application/views/user/login.php

Line Number: 36
```

StrangeWill, 2023-08-23:
Found a solution:

Changed my mounted volumes to only these files for the config directory:

```
      - ./volumes/config/config.php:/var/www/html/application/config/config.php
      - ./volumes/config/database.php:/var/www/html/application/config/database.php
```

(left the uploads and backup directory as per the docs)

Other configuration files change during upgrades, I also dropped the new `lang` section into the `config.php` file, not sure if this was required.

int2001, 2023-08-23:
The new lang-section at config.php is required for user-specific language settings. Without that there's only one lang-setting which sets the language instance-wide.