magicbug/Cloudlog #2430: A PHP Error was encountered

kieranlee999, 2023-08-25:
I have quite a few PHP errors listed on my cloudlog site. I've not been on for a while and all of a sudden it appears to not be working. 

Severity: 8192

Message: Return type of CI_Session_files_driver::open($save_path, $name) should either be compatible with SessionHandlerInterface::open(string $path, string $name): bool, or the #[\ReturnTypeWillChange] attribute should be used to temporarily suppress the notice

Filename: drivers/Session_files_driver.php

Line Number: 132

There are quite a few others as well [https://m6ayw.co.uk/Cloudlog](https://m6ayw.co.uk/Cloudlog)

phl0, 2023-08-25:
Which PHP version are you using?

kieranlee999, 2023-08-25:
So it was 8.2 but i've just dropped it down to 7.4 hoping that might fix it but still the same errors.

kieranlee999, 2023-08-25:
I'm now getting: 

There are multiple migrations with the same version number: 072.

magicbug, 2023-08-25:
What version of Cloudlog are you using? there's only 1 migration with 072

kieranlee999, 2023-08-25:
The version from: https://github.com/magicbug/Cloudlog/archive/master.zip

phl0, 2023-08-25:
> The version from: https://github.com/magicbug/Cloudlog/archive/master.zip

There is only one migration with 072 in the archive ...

[image]

Maybe switched between branches and some leftovers?

kieranlee999, 2023-08-25:
[image]
So There were 2 x 072 and 073 so I changed the oldest files to .old on the end now I have the error: 

A Database Error Occurred
Error Number: 1071

Specified key was too long; max key length is 1000 bytes

CREATE UNIQUE INDEX idx_paper_types_user_id_paper_name ON paper_types (user_id, paper_name) ALGORITHM DEFAULT LOCK DEFAULT;
Filename: migrations/134_create_label_paper_types_table.php
Line Number: 61

phl0, 2023-08-25:
It is hard to tell where the error is as it seems that there is some mixup of branches ... May also affect other files as well...

magicbug, 2023-08-25:
I'd suggest deleting the files in migration and replace with whatever is in the .zip and hope it fixes it

magicbug, 2023-08-25:
You can get the latest tagged release from https://github.com/magicbug/Cloudlog/archive/refs/tags/2.4.7.zip too

kieranlee999, 2023-08-26:
So i've redone the migrations folder but it now has the error: Too many redirects. It loads initially onto /Cloudlog/index.php/welcome then redirects to /Cloudlog/index.php/dashboard and then throws up the too many redirects.

petrkr, 2023-08-27:
> So i've redone the migrations folder but it now has the error: Too many redirects. It loads initially onto /Cloudlog/index.php/welcome then redirects to /Cloudlog/index.php/dashboard and then throws up the too many redirects.

This one I solving yesterday too

Just go by hand to https://<yourhostname>/user/login URL directly, actually this bug I do not know how to reproduce... I mentioned it yesterday in [ #2381](https://github.com/magicbug/Cloudlog/issues/2381#issuecomment-1694521379)

kieranlee999, 2023-08-27:
So i've got it working again. I've downgraded to 2.4.4 and it seems to be there and working again :)

petrkr, 2023-08-27:
> 

you not tried to go to that user/login url as I mention before?

magicbug, 2023-08-30:
Should hopefully find updating now resolves things

closed 2023-08-30 by magicbug