magicbug/Cloudlog #2435: New station location QSOs not showing in Logbook overview (or homepage) but does show in Advanced logbook view

M5JFS, 2023-08-26:
User error or a bug?

I created a new station location MU5JFS but QSO for 18.8.23 with TF/DL2GRC does not show in the Logbook overview
If I page pack in time, other station locations (eg. MR5JFS) do show in the overview.
The MU5JFS QSO does show in the advanced logbook view however and it also comes up in search results if I search for TF/DL2GRC

QSO showing in search results
[image]

QSO showing in logbook advanced view
[image]

QSO not showing in logbook overview
[image]

QSO not showing in homepage 
[image]

MR5JFS station location calls showing in logbook overview
[image]

magicbug, 2023-08-26:
I'm guessing when you created the new location you added it either to a new "Logbook" or an existing one? and that said ones active along with the location

M5JFS, 2023-08-26:
User error - I'd forgotten to add it to the list of locations for my main logbook 🤦‍♂️.

closed 2023-08-26 by M5JFS