magicbug/Cloudlog #2438: Problem matching LoTW record containing "DATA" as mode

lu1aax, 2023-08-26:
Hi, I am experiencing a problem with confirmation of JS8 qso. Record from LoTW comes with DATA as data mode and that seems to be the reason of not matching.

magicbug, 2023-08-26:
If you can provide a screenshot of a JS8 confirmed on LOtw website we can probably get to the bottom of it.

lu1aax, 2023-08-26:
Station
Call Sign	 	LU1AAX
DXCC	 	ARGENTINA (100)
CQ Zone	 	13
ITU Zone	 	14
Grid	 	GF05SJ
Worked Station
Worked	 	K4NXS
DXCC	 	UNITED STATES OF AMERICA (291)
CQ Zone	 	05
ITU Zone	 	08
Grid	 	EM95AC
State	 	South Carolina (SC)
County	 	Spartanburg
Date/Time	 	2023-08-26 20:11:40
Mode	 	DATA (DATA)
Band	 	10M
Frequency	 	28.07850
QSL	 	[2023-08-26 20:29:15](https://lotw.arrl.org/lotwuser/qsodetail?qso=1756870301)
Record ID 1756878972 Received: 2023-08-26 20:29:15

int2001, 2023-08-27:
JS8 is a "Submode" of MFSK, not of "DATA". That's the reason.

phl0, 2023-08-27:
The question is which mode and/or submode is contained in the Report that CL downloads from LoTW?

int2001, 2023-08-27:
@lu1aax can you Post two screenshots (as picture).

1.) from the lotw-page. QSL-Details
2.) out of Cloudlog. QSO-Details

And, if possible: an adif-export of only this QSO from Cloudlog

lu1aax, 2023-08-27:
Sure

[image]

[image]

[LU1AAX-27082023-1221.adi.txt](https://github.com/magicbug/Cloudlog/files/12447948/LU1AAX-27082023-1221.adi.txt)

lu1aax, 2023-08-27:
> The question is which mode and/or submode is contained in the Report that CL downloads from LoTW?

[image]

phl0, 2023-08-27:
I just created a test QSO in my CL dev environment. I chose JS8 as mode and then exported an ADIF file. This contains:
```
<MODE:4>MFSK
<SUBMODE:3>JS8
```
which I believe is correct. 
It seems to me that your QSO partner uploaded with different settings which result in DATA/DATA so the modes do not match on import. How do we know that DATA/DATA is JS8 and not something else? I guess K4NXS should check his log software.

lu1aax, 2023-08-28:
> I just created a test QSO in my CL dev environment. I chose JS8 as mode and then exported an ADIF file. This contains:
>
> ```
> <MODE:4>MFSK
> <SUBMODE:3>JS8
> ```
>
> which I believe is correct. It seems to [...]

Well, this was correctly matched in QRZ.com and LoTW, I would expect Cloudlog to do so...

phl0, 2023-08-31:
I am not sure what the issue is here. AFAIK LoTW does not take modes into calculation for matches because my previous cross-mode QSOs (e.g. me working CW / DX answering in SSB) did match. The LoTW report contains the mode that I sent during my LoTW upload. So it still would be nice to see what the lotwreport.adi contains when you download it from your LoTW account.

lu1aax, 2023-08-31:
> I am not sure what the issue is here. AFAIK LoTW does not take modes into calculation for matches because my previous cross-mode QSOs (e.g. me working CW / DX answering in SSB) did match. The LoTW [...]

I've attached the .adi file from LoTW as requested some days ago.

phl0, 2023-08-31:
Hmm strange. And you uploaded from your Cloudlog to LoTW directly?

lu1aax, 2023-09-04:
> Hmm strange. And you uploaded from your Cloudlog to LoTW directly?

Yes, I never upload to LoTW with other tools other than Cloudlog.