magicbug/Cloudlog #2443: Infinite redirect loop on new installation

petrkr, 2023-08-27:
Clear install causes redirect loops
it was already mentioned in #2381 and #2430

**To Reproduce**
1. Install fresh version
2. Go to install page
3. enjoy infinite looping

**Expected behaviour**
See maybe login page? Or maybe some install successful page?

**workaround**
go to url user/login by hand

---

I already found what is causes.. But do not know expected behavior of that function.

TL;dr
It is because of login regarding to 'version2_trigger'

Long version described in this comment
https://github.com/magicbug/Cloudlog/issues/2381#issuecomment-1694706619

magicbug, 2023-08-27:
PHP version number?

petrkr, 2023-08-27:
@magicbug 8.2
But regarding to code, that logic will be same in every version of PHP... Because how it is written it will do redirect to dashboard if user is not logged in, and dashboard will redirect it back, but user is still not logged in...

Or maybe, since my last PHP knowledge was around PHP5, I do not know if some evaluation of classes changed.. But I guess constructor is called before any class function for ages... But is right,, PHP was always unique by something... (but it is true,, still it is nothing compared to JavaScript, where true == true is actually false :) )

magicbug, 2023-08-28:
I'll take a look at it, really the upgrade stuff is only needed for old installs not the new ones

magicbug, 2023-08-28:
Perfect, while code might be the same its always helpful to have PHP version numbers just in case :)

magicbug, 2023-08-28:
OK so I think the solution to this is to add some SQL to the install setup so that the version_2 trigger is automatically set

```sql
INSERT INTO `options` (`option_id`, `option_name`, `option_value`, `autoload`) VALUES (NULL, 'version2_trigger', 'yes', 'true')
```

petrkr, 2023-08-28:
One of, but that option is already there as "false" from some migration script, so probably you will need UPDATE.

Or redone that logic with this flag. I do not know if you can recognize old database version and show that upgrade dialogue only if version is lower than 2.x and also when other stuff are met (that code which looking for api keys etc...)

magicbug, 2023-08-28:
OK actually made some more changes to the install.sql and the migration files which I hope will resolve the loop

petrkr, 2023-08-28:
Problem persists... 
probably this migration script will update it https://github.com/magicbug/Cloudlog/blob/master/application/migrations/102_add_version_two_trigger_to_options.php

magicbug, 2023-08-28:
Hmm, it shouldn't the script should only do something if version2_trigger isn't found which if its a fresh install it is there.

magicbug, 2023-08-28:
Yeah sorry, this is of course on the dev branch

petrkr, 2023-08-28:
Well ... progress.. there is no infinite loop, but new error occurred..

Maybe there should be semicolon after that new INSERT ?
Because it seems to not feed whole DB

```
An uncaught Exception was encountered
Type: mysqli_sql_exception

Message: Table 'devcloudlog.users' doesn't exist

Filename: /var/www/devcloudlog/system/database/drivers/mysqli/mysqli_driver.php

Line Number: 301
```

petrkr, 2023-08-28:
yes and also problem with SQL itself;

```
SQL query: [Copy](http://172.16.0.8/phpadmin/db_sql.php?db=devcloudlog#)

INSERT INTO `options` (`option_id`, `option_name`, `option_value`, `autoload`) VALUES (NULL, 'version2_trigger', 'true', 'yes')

MySQL said: Documentation
#1048 - Column 'option_id' cannot be null
```

petrkr, 2023-08-28:
well.. I'll do PR

petrkr, 2023-08-28:
yes, because also thta SQL error was because table missed indexes and auto increment, I compared it to production database and made it same

petrkr, 2023-08-28:
@HB9HIL can you please test it aswell ? So we can close this issue ?

HB9HIL, 2023-08-28:
> @HB9HIL can you please test it aswell ? So we can close this issue ?

ok hold on

HB9HIL, 2023-08-28:
> @HB9HIL can you please test it aswell ? So we can close this issue ?

Looks good 👍 well done

iu2frl, 2023-08-28:
Tried with PHP8.0-Apache on Docker, no more issues

commit fde3357d (2023-08-28): [Options] Creates options table during install and inserts the version2 flag and migration file makes sure it doesn't exisit.

This will hopefully resolve #2443 loop for new installs who of course don't need to run through the upgrade process for version 1 to 2
commit 5c7bcdc5 (2023-08-28): Install: Create auto-increment, fix semicolon, fixes #2443
commit 628bf0d2 (2023-08-28): Merge pull request #2448 from petrkr/fixinstaller

Install: Create auto-increment, fix semicolon, fixes #2443
closed 2023-08-28 by petrkr