magicbug/Cloudlog #2445: Feature Request: Refresh dashboard when new QSO is logged by API

petrkr, 2023-08-28:
If some software (like WSJT-X) log QSO, would be nice it will show up on dashboard as-well. It is almost same as #2060 but for QSO too.

But instead of HTTP Polling would be better to use web sockets, since every HTTP request especially on HTTPS have big overhead. For first it will make network load, but on other side it will mess access logs for every request. And if there will be more users, then such logs will be very huge.

magicbug, 2023-08-28:
Adding this using htmx like we do the radio status would be fine for this as it doesn't need to poll every second, in fact, we already do this on the QSO entry page for previous QSOs.

While websockets would be interesting, they would add an extra layer of complexity that isn't needed at the moment.

petrkr, 2023-08-28:
I just tried to read something about web socket server on PHP side. And it is right, PHP by design is stateless request-response language, so websocket is littlebit "hacked there".

I forgot all other web apps, which have WebSocket, have also backend in some Python, java, nodejs or other "daemon" based server, but not LAMP.

It is not need until lot of users will use one server or until users will complain battery usage on laptop/mobiles, since polling always communicate lot of data (ssl hanshake, request/response), but websocket only have TCP keepalives and only communicate when there are new data. Also for API it would be nice, if you tune radio, you can pretty fast send new frequencies, you do not need do thousands requests (but of course client can also send new frequency once per 10 sec and only if changed)

closed 2023-12-12 by magicbug