magicbug/Cloudlog #2453: QSL Request CSV contains unicode characters which MS Excel/MS Word doesn't like!

m1geo, 2023-08-30:
The Export to CSV button generates CSV that includes "Ø" characters which aren't correctly handled by Microsoft Excel/Word for using in mail merges. 

[image]

I know that Word does support these, but it gets broken when opened. Not sure if you can add some metadata or make this an option during the export.

[image]

Looking at the same QSOs in the ADIF shows a standard ASCII "0" character:

[image]

Just in case there is a way to make this import work better :)

Thanks for the great work,
George, M1GEO. sk sk.

phl0, 2023-08-30:
Fix proposed in https://github.com/magicbug/Cloudlog/pull/2454.

magicbug, 2023-08-30:
This code is now in the master branch so @m1geo should be able to use and test it easily.

closed 2023-10-26 by AndreasK79